realsenseai/librealsense #12571: Calibration problem in Realsense camera D435 1.20 meter

erayak, 2024-01-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                         |  D435 | 
| Firmware Version                     |  5.15.1  | 
| Operating System & Version  |  Ubuntu  22.04 LTS | 
| Kernel Version (Linux Only)    |  6.5.0-14-generic  | 
| Platform                                    |  PC  |
| SDK Version                             |  2.54.2.5684  | 
| Language                                 |  Python  | 
| Segment			          |  Others   | 

### Issue Description
Hello, we use https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam as a reference application. It calculates correctly from 60-70 cm, but we cannot be able to calculate from 1.20 meter. Our "HighResHighAccuracyPreset.json"  is like this;

```json
{
        "aux-param-autoexposure-setpoint": "1536",
        "aux-param-colorcorrection1": "0.298828",
        "aux-param-colorcorrection10": "-0",
        "aux-param-colorcorrection11": "-0",
        "aux-param-colorcorrection12": "-0",
        "aux-param-colorcorrection2": "0.293945",
        "aux-param-colorcorrection3": "0.293945",
        "aux-param-colorcorrection4": "0.114258",
        "aux-param-colorcorrection5": "-0",
        "aux-param-colorcorrection6": "-0",
        "aux-param-colorcorrection7": "-0",
        "aux-param-colorcorrection8": "-0",
        "aux-param-colorcorrection9": "-0",
        "aux-param-depthclampmax": "65536",
        "aux-param-depthclampmin": "0",
        "aux-param-disparityshift": "0",
        "controls-autoexposure-auto": "True",
        "controls-autoexposure-manual": "8500",
        "controls-color-autoexposure-auto": "True",
        "controls-color-autoexposure-manual": "166",
        "controls-color-backlight-compensation": "0",
        "controls-color-brightness": "0",
        "controls-color-contrast": "50",
        "controls-color-gain": "64",
        "controls-color-gamma": "300",
        "controls-color-hue": "0",
        "controls-color-power-line-frequency": "3",
        "controls-color-saturation": "64",
        "controls-color-sharpness": "50",
        "controls-color-white-balance-auto": "True",
        "controls-color-white-balance-manual": "4600",
        "controls-depth-gain": "16",
        "controls-laserpower": "360",
        "controls-laserstate": "on",
        "ignoreSAD": "0",
        "param-amplitude-factor": "0",
        "param-autoexposure-setpoint": "1536",
        "param-censusenablereg-udiameter": "9",
        "param-censusenablereg-vdiameter": "9",
        "param-censususize": "9",
        "param-censusvsize": "9",
        "param-depthclampmax": "65536",
        "param-depthclampmin": "0",
        "param-depthunits": "1000",
        "param-disableraucolor": "0",
        "param-disablesadcolor": "0",
        "param-disablesadnormalize": "0",
        "param-disablesloleftcolor": "0",
        "param-disableslorightcolor": "1",
        "param-disparitymode": "0",
        "param-disparityshift": "0",
        "param-lambdaad": "751",
        "param-lambdacensus": "6",
        "param-leftrightthreshold": "10",
        "param-maxscorethreshb": "2893",
        "param-medianthreshold": "796",
        "param-minscorethresha": "4",
        "param-neighborthresh": "108",
        "param-raumine": "6",
        "param-rauminn": "3",
        "param-rauminnssum": "7",
        "param-raumins": "2",
        "param-rauminw": "2",
        "param-rauminwesum": "12",
        "param-regioncolorthresholdb": "0.786693",
        "param-regioncolorthresholdg": "0.566536",
        "param-regioncolorthresholdr": "0.985323",
        "param-regionshrinku": "3",
        "param-regionshrinkv": "0",
        "param-robbinsmonrodecrement": "25",
        "param-robbinsmonroincrement": "2",
        "param-rsmdiffthreshold": "1.65625",
        "param-rsmrauslodiffthreshold": "0.71875",
        "param-rsmremovethreshold": "0.815476",
        "param-scanlineedgetaub": "13",
        "param-scanlineedgetaug": "15",
        "param-scanlineedgetaur": "30",
        "param-scanlinep1": "155",
        "param-scanlinep1onediscon": "160",
        "param-scanlinep1twodiscon": "59",
        "param-scanlinep2": "190",
        "param-scanlinep2onediscon": "507",
        "param-scanlinep2twodiscon": "493",
        "param-secondpeakdelta": "647",
        "param-texturecountthresh": "0",
        "param-texturedifferencethresh": "1722",
        "param-usersm": "1",
        "param-zunits": "1000"
}
```

We think it is due to calibration.
Do you have any suggestions for us to operate it properly at this height?

## 60cm
[image]

## 1.20m
[image]

MartyG-RealSense, 2024-01-12:
Hi @erayak  This issue with a diagonal bounding box in box_dimensioner_multicam.py has been encountered in a past case at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360044001733/comments/1500001054321

[image]

This tool has a 'sweet spot' distance for placing the camera of around half a meter from the checkerboard.  If the camera is moved further away from this though then problems such as the diagonal bounding box can occur.  There is not a known solution that allows a greater distance from the board, unfortunately.

erayak, 2024-01-13:
Thanks @MartyG-RealSense can we solve this problem with different realsense camera models and is there a different method other than chessboard as a calibration method?

MartyG-RealSense, 2024-01-13:
The distance issue will occur no matter which RealSense camera model is used with box_dimensioner_multicam

If you are using a single camera then box_dimensioner_multicam can be modified to be used without the chessboard if that camera's extrinsics are hard-coded into the program instead of being calculated in real-time using the chessboard.  Details about this can be found at https://github.com/IntelRealSense/librealsense/issues/9951#issuecomment-966039933

You could experiment with a larger chessboard with bigger square sizes to see whether it enables the program to work correctly when the camera is placed further away.  You could need to modify some values in the code to reflect the size changes made to the board though.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py#L40-L42

MartyG-RealSense, 2024-01-20:
Hi @erayak  Do you require further assistance with this case, please?  Thanks!

erayak, 2024-01-20:
Hi @MartyG-RealSense,

Can you clearify your last message more? Are you talking about specifying a measurement area and give the code x,y values in pixels and cm/mm.  Also should we use AI to detect the objects and then measure them. Would that increase the accuracy?

Thanks.

MartyG-RealSense, 2024-01-20:
Are you referring to my comment about a larger chessboard, please?  If you are then by this I mean that by default the program is set up to look at a chessboard image print-out on the floor that is 6x9 inches in size and squares on the chessboard image that have a size of 0.0253 meters or 2.53 centimeters.  This image can be obtained from the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/pattern_chessboard.png

So you can create and print a larger size chessboard image if you wish in order to try to improve detection over greater distance.  But if you make your own chessboard then you need to adjust the chessboard and board square size values in the script.

erayak, 2024-01-22:
Hi @MartyG-RealSense,

We are trying to remove chessboard calibration part of the program but as you know it uses kabsch algorithm and it returns transformation information. We can be able to set intrinsics and extrinsics but we also need transformation information. How should we proceed from this point?

Thanks.

closed 2024-02-05 by MartyG-RealSense