realsenseai/librealsense #12572: update pkg-config librealsense.pc.in file to support macos

mfbolus, 2024-01-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | **D400** | 
| Firmware Version                   | n/a | 
| Operating System & Version |   **MacOS**  | 
| Kernel Version (Linux Only)    |  n/a                                         | 
| Platform                                 | **PC** |
| SDK Version                            |  master branch                         | 
| Language                            |  {**C**}                          | 
| Segment			|  n/a                         | 

### Issue Description
Thank you all for your work. 

When linking against this library on MacOS using pkg-config, I discovered that the library search path argument is too specific to linux platforms. See the [librealsense.pc.in file](https://github.com/IntelRealSense/librealsense/blob/4673a37d981164af8eeb8e296e430fc1427e008d/config/librealsense.pc.in). Specifically, see the line `libdir= ${prefix}/lib/x86_64-linux-gnu`. On MacOS, this should just be `libdir= ${prefix}/lib`. I see a `TODO` note on the line above to address this issue. Is this something you all still have on your radar? 

Best, 
MFB

MartyG-RealSense, 2024-01-12:
Hi @mfbolus  The **librealsense_pc_in** file has not been updated since the original version of librealsense2 was being developed in 2017.

You could make the edit to the source code of **librealsense_pc_in** yourself and then build a custom version of librealsense from source for your particular project.  

This modification to **librealsense_pc_in** is not usually required to install librealsense on MacOS.

**Intel-based Macs**
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

**Apple Silicon M1 Macs**
https://github.com/IntelRealSense/librealsense/issues/11897#issuecomment-1584266362

mfbolus, 2024-01-13:
@MartyG-RealSense , thank you for the quick reply! I am using a binary installed via a package manager and linking against the library using cues from pkg-config. I understand mine is probably an edge case in MacOS, and I have a workaround that is fine for me. I just wanted to bring this up because it looked like it was something that might have been vestigial. 

Feel free to close this issue if you prefer.

Thanks again!

MartyG-RealSense, 2024-01-13:
Yes, it is a situation that does not usually arise.  It's great though that you shared your workaround for the benefit of Mac users who might need it in future.  Thanks so much!

closed 2024-01-13 by MartyG-RealSense