realsenseai/librealsense #12579: camrera vertical installation

Nimaro76, 2024-01-16:
Hello, 

Our team would like to install D435IF vertically and use them in that way. could you please guide us whether there are any side effects of such installation in terms of point clouds, depth frame and overall performance ? 

thank you for your guidance
kind regards

MartyG-RealSense, 2024-01-16:
Hi @Nimaro76  If you are only rotating the camera then there should be no negative consequences from rotating the camera 90 degrees on its side.

However, if you also plan to rotate the image 90 degrees so that its orientation is the 'right way up' then that would be problematic, as all calibration data and other parameters would need to be updated.

Nimaro76, 2024-01-16:
Thank you for your prompt response. Upon vertical installation, getting the depth image and obtaining point
clouds, then rotating the pointclouds 90 degrees around z axis, we observed a reduced FOV . is this logical ? 
kind regards

MartyG-RealSense, 2024-01-16:
I would expect some distortion of the pointcloud view when rotated instead of it simply being a perfect 90 degree flipped image of the original, yes.

MartyG-RealSense, 2024-01-24:
Hi @Nimaro76  Do you require further assistance with this case, please?  Thanks!

Nimaro76, 2024-01-24:
I have another question but not in this case that I would appreciate your guidance. I want to use a USB 3 hub to connect 3 comeras to my computer . could you please provide me with the information regarding the bandwidth requirement ? 
kind regards
Nima

MartyG-RealSense, 2024-01-24:
A USB hub typically has a maximum bandwidth of 5 GBPS (gigabits per second) that is shared between all devices attached to the hub.  The more simultaneous streams and the higher the FPS / resolution that each camera uses, the more of the total available bandwidth that will be consumed, so you need to ensure that there is enough bandwidth remaining for all attached cameras.

5 GBPS should be enough for 3 cameras.  If not then you could choose a hub with more bandwidth such as 10 GBPS or link two hubs together and put one of the 3 cameras on the second hub.  The USB standard allows up to 5 hubs to be 'daisy chained' together.

More information about hub bandwidth can be found at the link below.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#a-bandwidth

closed 2024-02-27 by MartyG-RealSense