realsenseai/librealsense #12583: Vision Processor D4 Board V3 doesn't work with D450 module and can't update firmware

emdc, 2024-01-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D450 | 
| Firmware Version                   | Unknown | 
| Operating System & Version |  Win11  | 
| Kernel Version (Linux Only)    |  -  | 
| Platform                                 | PC |
| SDK Version                            |  2.54.2 | 
| Language                            |  C++  | 
| Segment			| other | 

----------------------------------------------------------------------------------------------------

### Issue Description
I bought:
* Vision Processor D4 Board V3 (K94249-110)
* D450 module (K83121-101) 
* HSF100 cable.

I tried to connect to my PC and got:

```
> rs-fw-update.exe -l

Connected devices:
 17/01 16:40:58,863 WARNING [16740] (rs.cpp:312) null pointer passed for argument "device"
RealSense error calling rs2_create_device(info_list:000001A8AA5AF360, index:0):
    hr returned: HResult 0x80070492: "╨г╨║╨░╨╖╨░╨╜╨╜╤Л╨╣ ╨╜╨░╨▒╨╛╤А ╤Б╨▓╨╛╨╣╤Б╤В╨▓ ╨╜╨╡ ╤Б╤Г╤Й╨╡╤Б╤В╨▓╤Г╨╡╤В ╨┤╨╗╤П ╨╛╨▒╤К╨╡╨║╤В╨░."
```

Looks like the wrong message part is "The property set specified does not exist on the object." in Russian. 

Without D450 module I see in system devices list "Depth Camera 405i Tester", with D450 module - "wront enumeration".

I saw issues on GitHub (https://github.com/IntelRealSense/librealsense/issues/11621 and https://github.com/IntelRealSense/librealsense/issues/12444), so I tried to update the firmware without D450 module - got the same enumeration error. I tried to exec this command:
```
rs-fw-update.exe -f Signed_Image_UVC_5_15_1_0.bin
```

The firmware I downloaded from the official page https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads. 

What can I do in this situation?

MartyG-RealSense, 2024-01-18:
Hi @emdc  Thanks very much for posting your issue.  I will discuss it with my Intel RealSense colleagues.

MartyG-RealSense, 2024-02-20:
Hi @emdc  After my earlier email contacts with you in January, do you have an update about this case that you can provide please?  Thanks!

closed 2024-03-06 by MartyG-RealSense