realsenseai/librealsense #12584: D456 unity Slam scene error

yuhe0925, 2024-01-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D456 
| Firmware Version                   | 5.15.1
| Operating System & Version |  win10  
| Platform                                 | PC 
| SDK Version                            | 2.0                        
| Language                               |  unity                         

### Issue Description

Error message in SLAM scene on start play

ExternalException: rs2_pipeline_start_with_config(pipe:0000019C4225C9A0, config:0000019C4225D4A0)
Rethrow as Exception: Couldn't resolve requests
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <cb539c8577184d8bb3db94250e9792dc>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <cb539c8577184d8bb3db94250e9792dc>:0)
RsDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)

MartyG-RealSense, 2024-01-18:
Hi @yuhe0925  The Unity SLAM example only works with a now retired RealSense model called the T265 Tracking Camera.

MartyG-RealSense, 2024-01-26:
Hi @yuhe0925  Bearing in mind that the SLAM example only works with the RealSense T265 Tracking Camera, do you require further assistance with this case please?  Thanks!

closed 2024-02-02 by MartyG-RealSense