realsenseai/librealsense #12591: Not getting correct yaw when i mount realsense camera D455 on robot

adinathkale01, 2024-01-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux (Ubuntu 14/16/17) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC NVIDIA Gpu |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hi, I am using Realsense depth camera D455 for getting Imu data, when I check by holding the camera in hand it gives the correct Yaw, roll, and pitch but when I mount this camera on my robot it does not give correct values, I tried by rotating robot 90, 180,360 degrees still it is not giving correct, could you please explain to me how can I mount it. 
what I have to do to get the correct angles.

MartyG-RealSense, 2024-01-19:
Hi @adinathkale01  I believe that the accuracy of the IMU data may depend on how far away from the camera the location at which you are rotating the robot is.  For example, if the camera is on the front of the robot and you are rotating the robot at its center point, tilting the robot may result in a larger change in the camera's tilt angle than if you were rotating the camera in your hand directly.

If possible, moving the camera closer to the location on the robot at which you are rotating it (such as its center) may provide more correct results.

adinathkale01, 2024-01-19:
Thank you very much for responding.
we using the same Realsense camera also for detection purposes it should be attached to the front of the robot, could you suggest any mathematical(geometric)approach that I can use, to get the correct results?
In my case, it is a little difficult to mount the camera at the center, I have to mount it at the front for a better view.

MartyG-RealSense, 2024-01-19:
If you know the rough difference between the values when the camera is rotated by hand and when the whole robot is rotated, maybe you could add a calculation in the code that adjusts the gyro values once retrieved to take into account the distance of the camera from the robot's center.  For example, if you wanted to reduce the value by 10% then multiply the gyro angle by 0.90 (because 100% - 10% = 90%).

closed 2024-02-02 by MartyG-RealSense