realsenseai/librealsense #12595: Build failed on android examples

crdevvv, 2024-01-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   Win 10   |  
| Platform                                 | PC            |
| SDK Version                            |  2.0          | 
| Language                            |  Java           | 
| Segment			|  Smartphone     | 

### Issue Description
I'm trying to build the android modules following the description that you provide but it still outputs this this error:

Multiple build operations failed.
    Failed to notify dependency resolution listener.
    Failed to notify dependency resolution listener.
    Failed to notify dependency resolution listener.
    Failed to notify dependency resolution listener.
    Failed to notify dependency resolution listener.
    Failed to notify dependency resolution listener.
    Failed to notify dependency resolution listener.
    Failed to notify dependency resolution listener.
    Failed to notify dependency resolution listener.
Failed to notify dependency resolution listener.
'void org.gradle.api.artifacts.DependencySubstitutions$Substitution.with(org.gradle.api.artifacts.component.ComponentSelector)'

MartyG-RealSense, 2024-01-20:
Hi @crdevvv  This issue has not previously been reported in relation to RealSense.

My research of your question found a couple of Windows cases of the exact same problem where it was thought to have occurred after Android Studio had been upgraded.  Are you using Android Studio to build the RealSense Android wrapper, please?

If you are then one of these Windows users achieved a solution by uninstalling and then reinstalling Android Studio.

crdevvv, 2024-01-20:
Thanks for your answer @MartyG-RealSense.
Yes I'm using android studio for building android wrapper. So I'll try to uninstall and reinstall android studio then I'll give you an update.

MartyG-RealSense, 2024-01-28:
Hi @crdevvv  Do you have an update about this case that you can provide, please?  Thanks!

crdevvv, 2024-01-28:
Hi @MartyG-RealSense  sorry for the waiting, I tried to open the android module with another version (electric eel)of android studio and it works, so it's not a library building fault. 
Thanks for the support :)

MartyG-RealSense, 2024-01-28:
It's no problem at all.  It's great to hear that you achieved a working setup with Android Eel.  Thanks very much for the update!

closed 2024-02-05 by crdevvv