realsenseai/librealsense #12598: 下载时出现的问题

wangkangluyao, 2024-01-21:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:572

```

wangkangluyao, 2024-01-21:
怎么解决呢

MartyG-RealSense, 2024-01-21:
Hello @wangkangluyao  Please try using the instructions at https://github.com/IntelRealSense/librealsense/issues/10544#issuecomment-1138569631 to download and run the old RealSense Viewer version 2.21.0 and then closing it and launching the 2.54.2 viewer again to see whether the issue is corrected.

请尝试使用#10544（评论）中的说明下载并运行旧的 RealSense Viewer 版本 2.21.0，然后关闭它并再次启动 2.54.2 查看器，看看问题是否得到纠正。

MartyG-RealSense, 2024-01-26:
Hello @wangkangluyao  Do you require further assistance with this case, please?  Thanks!

您好@wangkangluyao 您需要有关此案的进一步帮助吗？ 谢谢！

MartyG-RealSense, 2024-02-02:
Case closed due to no further comments received.

由于没有收到进一步的评论，案件已结。

closed 2024-02-02 by MartyG-RealSense