realsenseai/librealsense #12599: Some error of realsense-viewer in Jetson NX

evallhq, 2024-01-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |  Linux - Ubuntu 18.04 | 
| Kernel Version (Linux Only)    | 4.9.201-tegra | 
| Platform                                 |  NVIDIA Jetson NX |
| SDK Version                            | 2.50.0 | 
| Language                            |  python| 
| Segment			|  Robot                     | 

### Issue Description
Hello,

I installed the Realsense SDK 2.0 in Jetson NX described as the Installing the packages part in the following document:
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

The whole installation appears successful -- I can run realsense_viewer with my realsense D455, but there are some following errors and mistakes: **(shown in the following figures)**

1. Jetson NX's USB interface is 3.0, but in the realsense-viewer it displays USB 2.1. I have looked up some other related **issues** and found that it may be due to **cable** issues. However, when I connected D455 to my PC using the same cable, the realsense-viewer displayed USB 3.2 (correct), so I am not sure what the problem is.
[image]

3. When I opened realsense-viewer everytime, there would show this error: 
```
Invalid Value in rs2_get_option(options:0x7(80003a70, option:Auto Gain Limit
Toggle):
hwmoh command 0x80 (5 0 0 0 ) failed (response -7= HW not ready)
```
[image]

I am very sorry that my Jetson NX has been embedded in a robot and cannot update its linux kernel (to prevent the robot system from crashing). I am very grateful that you can help me solve the above problems.

Thank you very much.

MartyG-RealSense, 2024-01-21:
Hi @evallhq  As you are using SDK version 2.50.0, the firmware driver version that is used should be **5.13.0.50**.  You have the newer firmware 5.15.1.0 installed in your camera.  When this firmware is used with SDK 2.50.0 it will cause the Auto Gain Limit Toggle error.  Downgrading your firmware to 5.13.0.50 should make the error disappear.

You can downgrade using the following procedure:

1.  Download the firmware .bin file for 5.13.0.50 from the firmware releases page at the link below.

https://dev.intelrealsense.com/docs/firmware-releases

2.  Open the RealSense Viewer and go to the **More** option near the top of the Viewer's options side-panel.  Select the **Update Firmware** option from its drop-down menu.  This will cause a file selector box to pop up.

3.  Navigate to the location on your Jetson where you downloaded the firmware .bin file to and select the file.  The firmware update process should then automatically begin.

Please let me know whether downgrading the firmware causes the USB 2.1 issue to be resolved.

evallhq, 2024-01-22:
> Hi @evallhq As you are using SDK version 2.50.0, the firmware driver version that is used should be **5.13.0.50**. You have the newer firmware 5.15.1.0 installed in your camera. When this firmware [...]

Thanks for your reply.

Yes, I have followed your idea and updated my firmware version to **5.13.0.50**. There have been no errors related to the Auto Gain Limit Toggle error. However, the USB interface still displays **2.1** instead of 3.0, and when I open the **motion module**, there will be an **Motion Module Faliure** error, as shown in the following figure:
[image]
When I connected D455 to my PC using the same cable, the realsense-viewer displayed USB 3.2 (correct), so I am still confused about this problem.

evallhq, 2024-01-22:
> > Hi @evallhq As you are using SDK version 2.50.0, the firmware driver version that is used should be **5.13.0.50**. You have the newer firmware 5.15.1.0 installed in your camera. When this firmware is used with SDK 2.50.0 it will cause the Auto Gain Limit Toggle error. Downgrading your firmware to 5.13.0.50 should make the error disappear.
> > You can downgrade using the following procedure:
> > 
> > 1. Download the firmware .bin file for 5.13.0.50 from the firmware releases page at the link below.
> > 
> > https://dev.intelrealsense.com/docs/firmware-releases
> > 
> > 2. Open the RealSense Viewer and go to the **More** option near the top of the Viewer's options side-panel.  Select the **Update Firmware** option from its drop-down menu.  This will cause a file selector box to pop up.
> > 3. Navigate to the location on your Jetson where you downloaded the firmware .bin file to and select the file.  The firmware update process should then automatically begin.
> > 
> > Please let me know whether downgrading the firmware causes the USB 2.1 issue to be resolved.
> 
> Thanks for your reply.
> 
> Yes, I have followed your idea and updated my firmware version to **5.13.0.50**. There have been no errors related to the Auto Gain Limit Toggle error. However, the USB interface still displays **2.1** instead of 3.0, and when I open the **motion module**, there will be an **Motion Module Faliure** error, as shown in the following figure: <img alt="image" width="722" src="https://private-user-images.githubusercontent.com/89585894/298507758-4168c475-cd99-400f-ac0f-d70dd403c08f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU5MTI2OTUsIm5iZiI6MTcwNTkxMjM5NSwicGF0aCI6Ii84OTU4NTg5NC8yOTg1MDc3NTgtNDE2OGM0NzUtY2Q5OS00MDBmLWFjMGYtZDcwZGQ0MDNjMDhmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTIyVDA4MzMxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRhYTEzZDRjYWYwMzRiY2NmMmUyYjdmMWE3YmNjYjQ5YzExNzViZTgxZmQ3OWEzNWQ3NDY2NmVmODBjZjhiNmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.na2u6wqvBkeKQS50RRFYAl4JLrV99Izd2tlTRno-z7U"> When I connected D455 to my PC using the same cable, the realsense-viewer displayed USB 3.2 (correct), so I am still confused about this problem.

I am very sorry that I have checked the USB port of my onboard Jetson NX, and it is 2.0. So maybe this is due to my own device. So you can only help me with the motion module failure error issue. 

Thank you very much.

evallhq, 2024-01-22:
I am very sorry that I have checked the USB port of my onboard Jetson NX, and it is 2.0. So maybe this is due to my own device. So you can only help me with the motion module failure error issue.

Thank you very much.

MartyG-RealSense, 2024-01-28:
Hi @evallhq  Do you require further assistance with this case, please?  Thanks!

evallhq, 2024-01-28:
> Hi @evallhq Do you require further assistance with this case, please? Thanks!

Thank you very much for your help. Because my embedded Jetson NX device has not been with me recently, I cannot continue checking for any issues. If there are any new situations in the future, I will open another issue to consult with you.

By the way, I have confirmed that the USB port of my NX is 2.0, and I would like to ask what impact this will have on me. For example, if I want to use D455 to connect this NX and call its depth image, will there be resolution and frame rate limitations if I only use Python to call its SDK (without ROS)?

Thank you again!

MartyG-RealSense, 2024-01-28:
You are very welcome!

Yes, the number of resolution and frame rate speeds supported by RealSense cameras on a USB 2 connection is limited compared to USB 3.  You can use 640x480 resolution at 30 frames per second on USB 2 though.  If you wanted to use 1280x720 resolution then on D455 it would be limited to 5 frames per second on USB 2.

evallhq, 2024-01-28:
Thanks for your reply.

So the USB port 2.0 will lead to resolution and frame rate limitations whether I use the python SDK or realsense-ros to use D455. I may need to make a choice between resolution and frame rate.

evallhq, 2024-01-28:
> You are very welcome!
>
> Yes, the number of resolution and frame rate speeds supported by RealSense cameras on a USB 2 connection is limited compared to USB 3. You can use 640x480 resolution at 30 [...]

Thanks for your reply.

So the USB port 2.0 will lead to resolution and frame rate limitations whether I use the python SDK or realsense-ros to use D455. I may need to make a choice between resolution and frame rate.

MartyG-RealSense, 2024-01-28:
Yes, resolution / FPS will be limited on USB 2 no matter whether you are using the Python pyrealsense2 wrapper, the RealSense ROS wrapper or any other interface, as the supported modes will be defined by the camera's firmware driver.

evallhq, 2024-01-28:
Got it. Thank you very much!

I have no question now and you can close this issue.

Thank you again!

MartyG-RealSense, 2024-01-28:
I'm pleased that I could help  :)

closed 2024-01-28 by MartyG-RealSense