realsenseai/librealsense #12601: pyrealsense2 support for Python 3.12

pasandal, 2024-01-22:
Hi,
Could you please provide information about your plans to support the latest Python version, 3.12. Last year, support for Python 3.11 was released in June, which was eight months after its official launch. I am interested in knowing whether it is standard practice for IntelRealSense to offer support eight months post the release of a new Python version. This information will help me to make decisions about our camera supplier. Thank you.

MartyG-RealSense, 2024-01-22:
Hi @pasandal  You should already be able to use Python 3.12 with the pyrealsense2 wrapper if you are able to compile the wrapper from source code instead of using **pip install pyrealsense2** to install from PyPi wheel packages.  A source code build of the wrapper can work with the latest Python versions that do not yet have PyPi packages.

The next release of the RealSense SDK is planned for around Q1 2024 (the first three months of the year).  The release of new PyPi packages coincides with SDK release dates because each new SDK version has its own PyPi packages.  

The latest Python version supported by the packages does not get upgraded with every release though.  For example, Python 3.8 and 3.9 support was added in SDK 2.44.0 and 3.11 support was added in 2.54.1.  So a source code build of the wrapper is the best option when you need to be able to use the latest Python versions quickly.

pasandal, 2024-01-22:
I am requesting pip install support for Python 3.12. In my case, using a source code wrapper is not feasible.

MartyG-RealSense, 2024-01-23:
Hi @pasandal  My colleagues plan to add PyPi package support for Python 3.12.  

If it cannot be implemented in time to be included in the next RealSense SDK release then they suggest for you a workaround of building the wrapper from source code and then uploading it as your own Python package until Intel can officially publish a package.  The link below has a guide for creating and uploading packages to PyPi.

https://www.turing.com/kb/how-to-create-pypi-packages

MartyG-RealSense, 2024-02-27:
Case closed due to no further comments received.

closed 2024-02-27 by MartyG-RealSense

DanielWang123321, 2024-03-27:
Also interesting in your plans to support the latest Python version 3.12 as I just got an error “Could not find a version that satisfies the requirement pyrealsense2” while using pip to install pyrealsense2. And I fully understood source code build is a better way than using pip install.

MartyG-RealSense, 2024-03-27:
At the time of writing this there are not pip wheel files available for 3.12.  They will likely be published when the next librealsense SDK version is released, which appears from the change logs in the **development** branch to be numbered '2.56'.

https://github.com/IntelRealSense/librealsense/tree/development

macsunmood, 2025-04-08:
Any updates on this ?

MartyG-RealSense, 2025-04-08:
Hi @macsunmood  Support for newer Python 3 versions is typically released at the same time as main releases of the librealsense SDK (the last main one was 2.55.1 and there have been a few beta versions since then).  The most recent estimate for release of the next main version - given on 28 January 2025 at https://github.com/IntelRealSense/librealsense/discussions/13572#discussioncomment-11985469 -  was 'start of Q2' - i.e sometime after the start of April.

macsunmood, 2025-04-08:
> Hi [@macsunmood](https://github.com/macsunmood) Support for newer Python 3 versions is typically released at the same time as main releases of the librealsense SDK (the last main one was 2.55.1 and there have been a few beta versions since then). The most recent estimate for release of the next main version - given on 28 January 2025 at [#13572 (reply in thread)](https://github.com/IntelRealSense/librealsense/discussions/13572#discussioncomment-11985469) - was 'start of Q2' - i.e sometime after the start of April.

thanks!

vstavros02, 2025-06-04:
Any further updates? This "pip install pyrealsense2" wont work on my Ubuntu 24.04 python 3.12 system

MartyG-RealSense, 2025-06-05:
Hi @vstavros02   Python 3.12 support was introduced in librealsense SDK beta version 2.56.1 onwards.

What you could try is installing beta **2.53.3**, as it has a pip install package for pyrealsense2 - `pyrealsense2-2.56.3.7838-cp310-cp310-manylinux1_x86_64_beta.whl` - that you could download to your computer and install after the 2.56.3 SDK has been installed.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.56.3

[image]

I believe that the downloaded .whl file can be pip installed with this commans:

pip install pyrealsense2-2.56.3.7838-cp310-cp310-manylinux1_x86_64_beta

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/13825 who also had Ubuntu 24.04 performed this procedure, though it did not work for them in their particular case.  The solution that worked for them was to install SDK 2.56.3 and then download the 2.55.1 version of pyrealsense2 from the link below and pip install with that instead.

If you try that then it should be sufficient just to use **pip install pyrealsense2** on 2.56.3 though, as the 2.55.1 version of pyrealsense2 is currently the latest available one on the online repository at the time of writing this.

https://pypi.org/project/pyrealsense2/#files