realsenseai/librealsense #12605: Why are the depths taken against a flat object by  (equivalent to the floor or wall) by D435 inconsistent?

birard, 2024-01-22:
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Win 10  | 
| SDK Version                            |  Intel RealSense Viewer v2.54.2 | 

[image]
why the depth of floor is different?
I want the background to have the same depth so I can use depth to separate the foreground and background.

MartyG-RealSense, 2024-01-22:
Hi @birard  First please try resetting the calibration of your D435 camera to its factory-new default calibration using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

birard, 2024-01-22:
I have calibrated according to the steps but the problem still exist. When i use a certatin distance and a angle, the depth of background will be consistent. And it needs to be very horizontal.  Some parts of the depth are correct, but it's hard to keep them all consistent.

birard, 2024-01-26:
@MartyG-RealSense Thanks ! Now, i have another question about the function colorizer. I want add a colorbar corresponding to depth and color so that i can see the depth value on the picture more directly. 
I am new to python, hope you can give me some ideas.

MartyG-RealSense, 2024-01-26:
If you wish to add a depth scale bar like the one at the edge of the RealSense Viewer depth image panel then that would be difficult in Python.  More information about this can be found at https://github.com/IntelRealSense/librealsense/issues/11570

MartyG-RealSense, 2024-02-02:
Hi @birard  Do you require further assistance with this case, please?  Thanks!

birard, 2024-02-04:
At present, I still haven't found a way to prevent the depth value from being affected by shadows. Is there any other solution that I can refer to?
My project is to use the depth value to scan the grain size as much as possible and separate it, but it seems that this is not a good method for small and complex scenes.

MartyG-RealSense, 2024-02-04:
If the camera is at a fixed height above the floor then you could try setting a **Threshold Filter** post-processing filter.  This filter enables you to set a minimum and maximum distance for depth rendering.  So by configuring the maximum depth distance that can be rendered on the depth image, you may be able to eliminate the shadowed floor from the image and leave the objects just above the floor remaining on the image.

closed 2024-02-20 by MartyG-RealSense