realsenseai/librealsense #12608: Unable to use bag file

Nikappa57, 2024-01-23:
----------------------------------------------------------------------
| Required Info                          |                                                                |
|----------------------------- |--------------------------------------  |
| Camera Model                        | D455                                                      | 
| Operating System & Version  |  MacOS 14.2.1                                        | 
| Platform                                  | macbook pro m1                                    |
| SDK Version                            |   2.54.2                                                  | 
| Language                                |  python                                                  | 
----------------------------------------------------------------------

### Unable to use bag file

I'm encountering an issue when attempting to use a bag file for testing my script. Every time I run the script, I encounter the following error:

```
pipeline.start(config)
RuntimeError: Couldn't resolve requests
```

code example:
```python
import pyrealsense2 as rs

...

pipeline = rs.pipeline()
config = rs.config()
rs.config.enable_device_from_file(config, args.input)
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 6)
pipeline.start(config)
```

This is my first attempt to use pyrealsense2 on my MacBook Pro M1. Despite exploring multiple solutions available online, I have not been able to resolve the issue. Any form of assistance or guidance would be greatly appreciated.

To install the python lib i used [pyrealsense2-macosx](https://github.com/cansik/pyrealsense2-macosx)

MartyG-RealSense, 2024-01-23:
Hi @Nikappa57  The FPS speed of '6' is not supported on the D455 camera model.  '5' should be used as the minimum FPS instead on D455.

Nikappa57, 2024-01-23:
Okay, thank you, I didn't know that. Anyway, I've already tried with different settings, both fps and resolution, and it doesn't change anything.  I tried setting it to 5 now, and it doesn't work.

MartyG-RealSense, 2024-01-23:
The resolution and FPS in the script should also match the resolution and FPS of the depth stream that is recorded in the bag file.

If you do not know the bag file's depth stream resolution and FPS then you can find these settings by opening the RealSense Viewer tool and playing the bag file back in the Viewer by drag and dropping the file into the center panel of the Viewer to automatically initiate playback, then looking in the side-panel for the stream settings by expanding open the 'Stereo Module' category.

[image]

[image]

Nikappa57, 2024-01-23:
My settings are:

```
{
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
     "stream-width": "1280"
}
```

I tried changing the resolution and fps because I found it as one of the possible solutions online, but I don't think that's the issue.
It's worth noting that on another Windows PC that we have always used, the code works.

MartyG-RealSense, 2024-01-23:
Is there any difference on your Mac if you specify the bag file name instead of args.input?  The bag file should be placed in the same directory as the script file.

For example, if the bag file was called test.bag:

`rs.config.enable_device_from_file(config, 'test.bag')`

Nikappa57, 2024-01-23:
I made a mistake, the complete configuration file for the recording is

```
{
    "device": {
        "fw version": "05.14.00.00",
        "name": "Intel RealSense D455",
        "product line": "D400"
    },
    "parameters": {
        "aux-param-autoexposure-setpoint": "1000",
        "aux-param-colorcorrection1": "-0.493164",
        "aux-param-colorcorrection10": "-0.272461",
        "aux-param-colorcorrection11": "-0.272461",
        "aux-param-colorcorrection12": "-0.355469",
        "aux-param-colorcorrection2": "0.831055",
        "aux-param-colorcorrection3": "0.831055",
        "aux-param-colorcorrection4": "-0.368164",
        "aux-param-colorcorrection5": "-0.133789",
        "aux-param-colorcorrection6": "-0.323242",
        "aux-param-colorcorrection7": "-0.323242",
        "aux-param-colorcorrection8": "1.19141",
        "aux-param-colorcorrection9": "0.90918",
        "aux-param-depthclampmax": "65536",
        "aux-param-depthclampmin": "0",
        "aux-param-disparityshift": "0",
        "controls-autoexposure-auto": "True",
        "controls-autoexposure-manual": "33000",
        "controls-color-autoexposure-auto": "True",
        "controls-color-autoexposure-manual": "156",
        "controls-color-backlight-compensation": "0",
        "controls-color-brightness": "0",
        "controls-color-contrast": "50",
        "controls-color-gain": "64",
        "controls-color-gamma": "300",
        "controls-color-hue": "0",
        "controls-color-power-line-frequency": "3",
        "controls-color-saturation": "64",
        "controls-color-sharpness": "50",
        "controls-color-white-balance-auto": "True",
        "controls-color-white-balance-manual": "4600",
        "controls-depth-gain": "16",
        "controls-laserpower": "150",
        "controls-laserstate": "on",
        "ignoreSAD": "0",
        "param-amplitude-factor": "0.08",
        "param-autoexposure-setpoint": "1000",
        "param-censusenablereg-udiameter": "9",
        "param-censusenablereg-vdiameter": "9",
        "param-censususize": "9",
        "param-censusvsize": "9",
        "param-depthclampmax": "65536",
        "param-depthclampmin": "0",
        "param-depthunits": "1000",
        "param-disableraucolor": "0",
        "param-disablesadcolor": "0",
        "param-disablesadnormalize": "0",
        "param-disablesloleftcolor": "0",
        "param-disableslorightcolor": "0",
        "param-disparitymode": "0",
        "param-disparityshift": "0",
        "param-lambdaad": "800",
        "param-lambdacensus": "26",
        "param-leftrightthreshold": "24",
        "param-maxscorethreshb": "2047",
        "param-medianthreshold": "500",
        "param-minscorethresha": "1",
        "param-neighborthresh": "7",
        "param-raumine": "1",
        "param-rauminn": "1",
[... 115 lines cut ...]
        # Convert depth_frame to numpy array to render image in opencv
        depth_color_image = np.asanyarray(depth_color_frame.get_data())

        # Render image in opencv window
        cv2.imshow("Depth Stream", depth_color_image)
        
        # Color frame
        color_frame = frames.get_color_frame()
        color_image = np.asanyarray(color_frame.get_data())
        cv2.imshow("Color Stream", color_image)
		

        key = cv2.waitKey(1)
        # if pressed escape exit program
        if key == 27:
            cv2.destroyAllWindows()
            break

finally:
    pass
```

[image]

Is this normal behavior?

MartyG-RealSense, 2024-01-23:
If a config instruction to define the resolution, format and FPS is not specified then it is likely playing back every stream that is stored in the file simultaneously and using the resolution / FPS that the streams were recorded at.  This is similar to playback in the RealSense Viewer, where playback automatically displays all streams contained in the bag at the stored resolution / FPS without having to specify settings.  

So whilst it is common to provide stream config instructions in a bag playback script, there is nothing wrong with your approach either.

Nikappa57, 2024-01-23:
Ok, all clear, thank you. However, is there any other way to resolve the earlier error? Otherwise, I'll proceed without configuring the stream.

MartyG-RealSense, 2024-01-23:
There have been past cases where code that works fine on a windows / Linux PC has problems on a non-PC computer / computing board for reasons that are not clear.  I cannot see any obvious problems in your code that would cause the **RuntimeError: Couldn't resolve requests** error.  

This error is usually caused because a stream configuration has been requested that the camera is unable to provide.  But you are retrieving the streams from a bag file, so a camera-related issue could be ruled out.  I note that the code works correctly on a Windows PC, which indicates that the code is okay and something else is causing the issue with the config instructions on your Mac.

I think that if the stream that is stored in the bag is recorded in the same resolution / FPS that you want to define in config instructions in your script, then you could proceed without configuring the streams.

Nikappa57, 2024-01-24:
Thank you very much for the quick and helpful assistance you provided. I'll continue on my way.

closed 2024-01-24 by Nikappa57