realsenseai/librealsense #12612: how to correct record bag file?

diplomatist, 2024-01-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 2.53.1 | 
| Operating System & Version |   Ubuntu20.04 LTS  | 
| Kernel Version (Linux Only)    |  (e.g. 5.10.110)                                         | 
| Platform                                 | rockchip rk3588s  |
| SDK Version                            |  2.53.1 }                          | 
| Language                            | c++                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
`	rs2::config cfg;;
	rs2::pipeline pipe;
	cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30);
	cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
	//cfg.enable_record_to_file("/home/firefly/realsense_record_test/recorder/test2.bag");
	//cfg.enable_record_to_file("/home/firefly/realsense_record_test/recorder/test1.bag");
	cfg.enable_record_to_file("/home/firefly/realsense_record_test/recorder/test.bag");
	pipe.start(cfg);
	for (int i = 0; i < 121; i++) {
		auto frames = pipe.wait_for_frames();
	}
	pipe.stop();
`
By executing the above code, I obtained three different sizes of bag files.
[image]

`	std::cout << "read bag" << std::endl;
	rs2::config cfg;
	cfg.enable_device_from_file("/home/firefly/realsense_record_test/recorder/test.bag");
	rs2::pipeline pipe;
	pipe.start(cfg);
	for (int i = 0; i < 120; i++)
	{
		rs2::frameset frames;
		pipe.poll_for_frames(&frames);
		std::cout << "pop frames" << std::endl;
		rs2::frame color_frame = frames.get_color_frame();
		Mat img(Size(640, 480), CV_8UC3, (void*)color_frame.get_data(), Mat::AUTO_STEP);
		imwrite("/home/firefly/realsense_record_test/recorder/test.png", img);
	}
	pipe.stop();
`

By executing the above code：
> read bag
pop frames
terminate called after throwing an instance of 'rs2::error'
  what():  null pointer passed for argument "frame_ref"
Aborted (core dumped)

I tried to read the saved bag file using viewer.exe, but encountered errors:
① Program terminated due to an unrecoverable SEH exception: Access Violation!
② Bag unisdered

Does this seem like pipe. stop() was not executed correctly?

MartyG-RealSense, 2024-01-27:
Hi @diplomatist  A **bag unindexed** error usually indicates that the bag recording was not properly closed, preventing the bag from being played back afterwards.

Some RealSense users pause the recording before stopping the pipeline.  Intel's **record-playback** C++ example page describes how to pause recording with `device.as<rs2::playback>().pause()`

https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback

diplomatist, 2024-01-29:
I found that rk3588s uses version 2.53.1 SDK, but PC uses version 2.54.2 SDK.
After downgrading the SDK version of the PC to 2.53.1, I can now read the recorded packages.

c++ record code
```c++
// c++ record code
	auto pipe = std::make_shared<rs2::pipeline>();
	pipe->start();
	std::cout << "sleep 1s warmup" << std::endl;
	sleep(1);
	pipe->stop();
	pipe = std::make_shared<rs2::pipeline>();
	rs2::config cfg;
	cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30);
	cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
	cfg.enable_record_to_file("/home/firefly/rs-record-bag/recorder/test.bag");
	pipe->start(cfg);
	rs2::device device = pipe->get_active_profile().get_device();
	for (int i = 0; i < 60; i++) {
		pipe->wait_for_frames();
	}
	device.as<rs2::recorder>().pause();
	pipe->stop();
	pipe = std::make_shared<rs2::pipeline>();
```

c++ read bag code
```c++
	auto pipe = std::make_shared<rs2::pipeline>();
	rs2::config cfg;
	cfg.enable_device_from_file("/home/firefly/rs-record-bag/recorder/test.bag");
	pipe->start(cfg);
	rs2::frameset frames;
	while (true) {
		frames = pipe->wait_for_frames();
		std::cout << "pop frames" << std::endl;
		rs2::frame color_frame = frames.get_color_frame();
		Mat img(Size(640, 480), CV_8UC3, (void*)color_frame.get_data(), Mat::AUTO_STEP);
		imwrite("/home/firefly/rs-record-bag//recorder/test.png", img);
		imshow("frame", img);
		waitKey(0);
	}
	pipe->stop();
	pipe = std::make_shared<rs2::pipeline>();
```
The above code is for testing purposes only.

I use `wait_ For_ Frames() `to read the frame stream of the bag file; I use` pipe ->poll_ For_ Frames (&frames)` will encounter the issue of `null points "frame_ref"`, which is different from the official provided reading bag example. It is unclear what the problem is.

MartyG-RealSense, 2024-01-29:
Hi @diplomatist  This sounds like an issue that was already reported in https://github.com/IntelRealSense/librealsense/issues/12213 where a bag created with 2.53.1 could be played back in that version but not in newer versions.  A bug report has been created about this issue.

MartyG-RealSense, 2024-02-04:
Hi @diplomatist  Do you require further assistance with this case, please?  Thanks!

diplomatist, 2024-02-04:
> Hi @diplomatist Do you require further assistance with this case, please? Thanks!

Thanks for your help!

closed 2024-02-04 by diplomatist