realsenseai/librealsense #12613: Why can‘t I use D435 in arm docker for Jetson Nano？

GuoPingPan, 2024-01-27:
All the sdk complie had passed but can not get any frames.
```
docker run -d --env="DISPLAY" --env="QT_X11_NO_MITSHM=1" 
--volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" 
--volume="/home/unitree/realsense_sender.py:/home/tealab/realsense_sender.py" 
--privileged=true --volume="/dev:/dev" 
--device-cgroup-rule "c 81:* rmw" 
--device-cgroup-rule "c 189:* rmw" 
--name controller --net=host controller:v7 tail -f /dev/null
```

[image]

MartyG-RealSense, 2024-01-27:
Hi @GuoPingPan  Does it work if you use **sudo docker run** instead of just 'docker run' please?

`sudo docker run -d --env="DISPLAY" --env="QT_X11_NO_MITSHM=1" `

Elgce, 2024-01-27:
Hi @MartyG-RealSense  I am cooperating with @GuoPingPan , and I have tried sudo, but it turns out to be useless

MartyG-RealSense, 2024-01-27:
There was a similar case with Docker and Nano at https://github.com/IntelRealSense/librealsense/issues/12218 where trying to run a Python script also resulted in **RunTimeError: Frame didn't arrive within 5000**.

In that case I suggested attempting to reload the **udev** device handling rules.

```
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control –reload-rules && udevadm trigger
```

GuoPingPan, 2024-01-28:
Thanks for your advices. I am tring to complie from source on `raspberrypi 4B` or `jetson nano`, but it will both struck in `Building CXX object CMakeFiles/realsense2.dir/common/utilities/time/l500/get-mfr-ww.cpp.o`. Why?

GuoPingPan, 2024-01-28:
What's more, in order to deal with the situation that my device(raspberrypi 4B or jetson nano) can't connect to internet, so I pre-clone pybind to `build/third-party` and uncomment th `ExternalProject_Add` in [https://github.com/IntelRealSense/librealsense/blob/master/CMake/pybind11-download.cmake.in](https://github.com/IntelRealSense/librealsense/blob/master/CMake/pybind11-download.cmake.in).

Will it cause any problem?

MartyG-RealSense, 2024-01-28:
I could not find a reason for the **get-mfr-ww** error, unfortunately.

When building from source, are you using **DFORCE_RSUSB_BACKEND=TRUE** please?  RSUSB relies on installing dependencies over an internet connection.  Applying a kernel patch and not using RSUSB = true when building from source may therefore work better for you, as a non-RSUSB source code build is not dependent on an internet connection.

GuoPingPan, 2024-01-28:
Oh sorry, I mean it just stop at `get-mfr-ww` not clash. What is the difference between `DFORCE_RSUSB_BACKEND=TRUE` and `DFORCE_RSUSB_BACKEND=FALSE`?

MartyG-RealSense, 2024-01-28:
RSUSB = false is called a 'native build' of the SDK and is the default backend of librealsense.  It is reliant on a kernel patch script being applied to the kernel.  An RSUSB = true build is not dependent on Linux versions or kernel versions and does not require kernel patching, but needs an internet connection to install dependencies.

You can find a detailed comparison of a kernel-patched build versus non-patched by going to https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604 and scrolling down through the linked-to comment to the section headed **What are the advantages and disadvantages of using libuvc vs patched kernel modules?**

GuoPingPan, 2024-01-29:
Thank you very much for your reply! I found that even though I was not connected to the network, I could compile the entire realsense project from source code with ` RSUSB = true`. It can also be used eventually.