realsenseai/librealsense #12614: fatal: You need to specify a tag name (Ubuntu 20.04, 5.15.76-rt53)

upc-ghy, 2024-01-27:
I followed the installation tutorial at https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md to install my librealsense, but encountered the following issues. Please help!

Ubuntu 20.04

**unmae -r**
5.15.76-rt53

**./scripts/patch-realsense-ubuntu-lts-hwe.sh**

Reading package lists... Done
Building dependency tree      
Reading state information... Done
bc is already the newest version (1.07.1-2build1).
build-essential is already the newest version (12.8ubuntu1.1).
git is already the newest version (1:2.25.1-1ubuntu3.11).
linux-headers-generic is already the newest version (5.4.0.170.168).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-focal folder

Package required libelf-dev:  - found
Package required elfutils:  - found
Package required bison:  - found
Package required flex:  - found
Package required dwarves:  - found
Fetching Ubuntu LTS tag   to the local kernel sources folder
fatal: You need to specify a tag name.

MartyG-RealSense, 2024-01-27:
Hi @upc-ghy  Are you installing on a PC or on a computing board such as Raspberry Pi, please?

upc-ghy, 2024-01-28:
@MartyG-RealSense I am using a PC with a 1080Ti.

MartyG-RealSense, 2024-01-28:
It is not clear why you would be experiencing the **fatal: You need to specify a tag name** error on a PC, as it usually occurs on computing devices such as Raspberry Pi.

Do you have the option of building librealsense from packages instead of source code, please?  The packages have the kernel patch bundled inside them, so you do not need to run the **./scripts/patch-realsense-ubuntu-lts-hwe.sh** patch script.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

upc-ghy, 2024-01-28:
@MartyG-RealSense Yes, I can install using this tutorial https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md without any errors. However, when I use the command "modinfo uvcvideo | grep 'version:'", I found that it does not contain the string "realsense".

**modinfo uvcvideo | grep "version:"**
```
version:        1.1.1
srcversion:     B5BC32E2A05C003B6787C94

```

MartyG-RealSense, 2024-01-28:
There was a recent case of this problem with no RealSense string at https://github.com/IntelRealSense/librealsense/issues/12581 and my advice was that as the Clock Domain was showing as **global time** in the RealSense Viewer tool, their kernel was likely okay despite the absence of the RealSense string because if the kernel was not patched then the Clock Domain would show as **system time** instead.

However, there have been past problems with using real-time kernels with librealsense - as described by an Intel RealSense team member at https://github.com/IntelRealSense/librealsense/issues/2395#issuecomment-426536906 - so this could be the cause of your problem.

upc-ghy, 2024-01-28:
@MartyG-RealSense Thanks for your reply.
I will test building librealsense from packages again, hoping it will work as you described.

upc-ghy, 2024-01-29:
@MartyG-RealSense 

I just test building librealsense from packages again. I encountered the following problem this time. 
I used command  `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge` to remove all realsense packages, referred to #11521  before reinstalling, but still encountered this problem.

RROR (dkms apport): kernel package linux-headers-5.15.76-rt53 is not supported

Error! Bad return status for module build on kernel: 5.15.76-rt53 (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.24/build/make.log for more information.
dpkg: error processing package librealsense2-dkms (--configure):
 installed librealsense2-dkms package post-installation script subprocess returned error exit status 10
Setting up librealsense2-utils:amd64 (2.54.2-0~realsense.10772) ...
Processing triggers for libc-bin (2.31-0ubuntu9.14) ...
Errors were encountered while processing:
 librealsense2-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

MartyG-RealSense, 2024-01-29:
It is likely the realtime nature of this kernel version that is being used that is the problem, since kernel 5.15 is officially supported by librealsense via patch **./scripts/patch-realsense-ubuntu-lts-hwe.sh**

If your project requires the realtime kernel and librealsense cannot work with this kernel with either a package or source code build then the remaining option for resolving it would be to build from source code with the flag **-DFORCE_RSUSB_BACKEND=TRUE** included in the CMake build instruction.  An RSUSB = true build of librealsense bypasses the kernel and so is not dependent on Linux versions or kernel versions and does not require kernel patching.

closed 2024-02-13 by MartyG-RealSense