realsenseai/librealsense #12617: I can not 'make' this project.

cabbage404, 2024-01-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Operating System & Version |   {Win 10  |                  |                | 

### Issue Description
When I 'make' the project, I encountered the following problem. How do I solve it?

In file included from D:/pythonDemo/librealsense-master/third-party/rsutils/src/dispatcher.cpp:6:0:
D:/pythonDemo/librealsense-master/third-party/rsutils/include/rsutils/time/waiting-on.h: In instantiation of 'void rsutils::time::waiting_on<T>::wait_until(const U&, const L&) [with U = std::chrono::duration<long long int, std::ratio<1ll, 1000000000ll> >; L = dispatcher::flush(std::chrono::_V2::steady_clock::duration)::<lambda()>; T = bool]':
D:/pythonDemo/librealsense-master/third-party/rsutils/src/dispatcher.cpp:117:76:   required from here
D:/pythonDemo/librealsense-master/third-party/rsutils/include/rsutils/time/waiting-on.h:162:53: error: 'class rsutils::time::waiting_on<bool>::wait_state_t' has no member named '_m'
         std::unique_lock< std::mutex > locker(_ptr->_m);
                                               ~~~~~~^~
D:/pythonDemo/librealsense-master/third-party/rsutils/include/rsutils/time/waiting-on.h:163:15: error: 'class rsutils::time::waiting_on<bool>::wait_state_t' has no member named '_cv'
         _ptr->_cv.wait_for( locker, timeout, [&]() -> bool {
         ~~~~~~^~~
third-party/rsutils/CMakeFiles/rsutils.dir/build.make:106: recipe for target 'third-party/rsutils/CMakeFiles/rsutils.dir/src/dispatcher.cpp.obj' failed
make[2]: *** [third-party/rsutils/CMakeFiles/rsutils.dir/src/dispatcher.cpp.obj] Error 1
CMakeFiles/Makefile2:874: recipe for target 'third-party/rsutils/CMakeFiles/rsutils.dir/all' failed
make[1]: *** [third-party/rsutils/CMakeFiles/rsutils.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2

MartyG-RealSense, 2024-01-29:
Hi @cabbage404  Most Windows users do not need to use 'make' to build librealsense as there is an installer program to install it automatically with just a few clicks.  This **Intel.RealSense.SDK-WIN10** installer can be downloaded and run from the 'Assets' file list at the bottom of the page at the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

[image]

The installer includes Python support in the components that it installs.

[image]

cabbage404, 2024-01-29:
> Hi @cabbage404 Most Windows users do not need to use 'make' to build the librealsense SDK from source code with CMake as there is an installer program to install it automatically with just a few clicks. This **Intel.RealSense.SDK-WIN10** installer can be downloaded and run from the 'Assets' file list at the bottom of the page at the link below.
> 
> https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2
> 
> ![image](https://private-user-images.githubusercontent.com/41145062/300437624-55bb271b-0bfa-42fa-9f13-ddae075ca588.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY1Mjg4NzIsIm5iZiI6MTcwNjUyODU3MiwicGF0aCI6Ii80MTE0NTA2Mi8zMDA0Mzc2MjQtNTViYjI3MWItMGJmYS00MmZhLTlmMTMtZGRhZTA3NWNhNTg4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI5VDExNDI1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVjZWI4NTlkODM3ODc2ODI3Zjg1Y2U2NTcyMDdmNGY0NjA4ZWM1OWY4NzExMDBmYTBiMTFiNzE1ODJjNGMwNTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.d4JRI7rsUzkl3O5YxAQ-VZJTiIRW8Qi8efUKaECMPLI)
> 
> The installer includes Python support in the components that it installs.
> 
> ![image](https://private-user-images.githubusercontent.com/41145062/300438601-dbbc2219-7ddd-4a56-a939-f8c67604f981.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY1Mjg4NzIsIm5iZiI6MTcwNjUyODU3MiwicGF0aCI6Ii80MTE0NTA2Mi8zMDA0Mzg2MDEtZGJiYzIyMTktN2RkZC00YTU2LWE5MzktZjhjNjc2MDRmOTgxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI5VDExNDI1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNhYzMyZmVkNTcwYWJhOTA1Y2RiZTY2YzdjYzI4ZmFiNzdhOWQxN2U4MDExYTA2MjlmM2YxNjI2M2Y2ZWI2ZDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4yTZiCaOcVKuOtaYu2J3yR__VYUSCxU2SA9WUIgQZL8)
I have downloaded this SDK and I would like to convert the. bag format to images or videos. It seems that this feature is not available in the SDK.

MartyG-RealSense, 2024-01-29:
Once the SDK has been installed on your Windows computer, it has a program called **rs-convert.exe** in its 'Tools' folder that can convert bag file frames to PNG image files.  

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

You can find the Tools folder easily by right-clicking on the **RealSense Viewer** launch shortcut icon that the installer places on the Windows desktop and selecting the 'Open file location' option from the menu.

rs-convert must be run in the Windows **Command Prompt** text interface, which can be opened by typing the word 'command' into the search box at the bottom of your Windows screen..  Instructions for launching rs-convert in the Command Prompt  can be found at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360051132573/comments/360014080493

The rs-convert tool is not able to convert bag files to video though, however.

MartyG-RealSense, 2024-01-30:
In the same Tools folder is an example program called **rs-save-to-disk.exe** that can capture a single live depth and color frame as a PNG image.

https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk

cabbage404, 2024-01-30:
> In the same Tools folder is an example program called **rs-save-to-disk.exe** that can capture a single live depth and color frame as a PNG image.
> 
> https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk

Thank you very much. May I ask how to use Python to obtain real-time depth and image streams?

MartyG-RealSense, 2024-01-30:
The Python example program **opencv_viewer_example.py** may meet your needs.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py

[image]

closed 2024-02-13 by MartyG-RealSense