realsenseai/librealsense #12620: Question about HW Sync using D455

FMatsusaki, 2024-01-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.15.1.0 | 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.54.2                   | 
| Language                            | python                       | 
| Segment			|  Robot                          | 

### Issue Description
[question]

I am trying the D455 multi camera(2 devices)  while reading the following doc.
https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

Now, I have a question.
Is D455 support Depth and RGB HW sync? There are multiple pieces of information below, and I am confused.

 In looking at some issues, it was mentioned that RGB HW sync is not supported.(https://github.com/IntelRealSense/librealsense/issues/12321#issuecomment-1780932585)
But, In DataSheet of D400 series, following information is written.

> 3.7.4 External Sensor Sync Connector
> For D415 and D455, both the depth and RGB sensors can be synced by an external trigger

MartyG-RealSense, 2024-01-30:
Hi @premQee  In theory, using Inter Cam Sync Mode '3' (Full Slave) on both cameras is supposed to enable depth + RGB hardware sync.  In practice though, this mode did not work well.  It was also designed for use with the D415 camera model.  This mode, part of the **genlock** external hardware sync system, is no longer supported by Intel and modes 1 and 2 (Master and Slave) are the only officially supported and validated Inter Cam Sync Modes.  However, the genlock system's modes including '3' are still usable in the RealSense SDK.

Whilst you are welcome to try mode '3', I would strongly recommend assuming that only depth will be able to be synced.

premQee, 2024-01-31:
Thank you for your response. 
In that case, does it mean that I can't check the (color) synchronization using the high-speed LED panel in the document?

> In theory, using Inter Cam Sync Mode '3' (Full Slave) on both cameras is supposed to enable depth + RGB hardware sync.

Is this correct? I thougaht one device is '1' (Master) and another is '3' (Full Slave)

MartyG-RealSense, 2024-01-31:
Mode 3 (Full Slave) will be set on both cameras if an external hardware sync trigger pulse is being used. 

If you wish to use one of your two cameras as a Master though so that it generates the sync trigger pulse, then the Master camera will be set as '1' and the Full Slave camera set as '3'.

premQee, 2024-01-31:
I understand. Can I do this test?
> In that case, does it mean that I can't check the (color) synchronization using the high-speed LED panel in the document?

MartyG-RealSense, 2024-02-01:
The LED panel product highlighted in the sync paper was for depth sync testing.  That paper was written years before genlock sync was introduced, so there is no past example of it being used with RGB.

In a past investigation, I identified the LED panel product in the paper as very likely being the one at the link below.

https://www.image-engineering.de/products/equipment/measurement-devices/900-led-panel

premQee, 2024-02-01:
OK. I will check "timestamp drrift" when hw sync is enabled.

MartyG-RealSense, 2024-02-09:
Hi @premQee  Do you have an update about this case that you can provide, please?  Thanks!

premQee, 2024-02-13:
I'm sorry for the delay.
I have confirmed the "timestamp drrift" using 2 devices. (Inter Cam Sync Mode is 1 and 2)
I can close this issue. @MartyG-RealSense  Thank you for your support.

MartyG-RealSense, 2024-02-13:
You are very welcome, @premQee - thanks very much for the update.  As you suggested, I will close this issue.  Thanks again!

closed 2024-02-13 by MartyG-RealSense