realsenseai/librealsense #12623: Is there a way to get imu data in docker?

ashBabu, 2024-01-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D456 | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  6.5.0-15                                         | 
| Platform                                 | docker  |
| SDK Version                            |  2.54.1                          | 

### Issue Description
This [docker](https://github.com/IntelRealSense/librealsense/blob/master/scripts/Docker/readme.md) building mentions that IMU cannot be accessed in docker. I was wondering if any further changes have been made to get IMU on docker since the link was updated around 3 years back.

I ran `ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true enable_gyro:=true enable_accel:=true unite_imu_method:=2` in docker

Running the above straight away throws error. I would like to get the /camera/imu topic.

```shell
662 ERROR [139936804103744] (librealsense-exception.h:52) Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1.1/4-1.1:1.5/0003:8086:0B5C.003C/HID-SENSOR-200073.7.auto/iio:device1/scan_elements/in_accel_y_en Last Error: Read-only file system
[realsense2_camera_node-1]  30/01 23:18:37,663 ERROR [139936804103744] (backend-hid.cpp:1005) Hid device is busy!
[realsense2_camera_node-1] [ERROR] [1706656717.663373915] [camera.camera]: ./src/rs_node_setup.cpp:370:An exception has been thrown: Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1.1/4-1.1:1.5/0003:8086:0B5C.003C/HID-SENSOR-200073.7.auto/iio:device1/scan_elements/in_accel_y_en Last Error: Read-only file system
[realsense2_camera_node-1] Failed to start device: Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1.1/4-1.1:1.5/0003:8086:0B5C.003C/HID-SENSOR-200073.7.auto/iio:device1/scan_elements/in_accel_y_en Last Error: Read-only file system
[realsense2_camera_node-1] [INFO] [1706656717.664063767] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1706656717.667175929] [camera.camera]: Starting Sensor: Motion Module
[realsense2_camera_node-1]  30/01 23:18:37,667 WARNING [139936804103744] (backend-hid.cpp:1205) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto
[realsense2_camera_node-1]  30/01 23:18:37,667 WARNING [139936804103744] (backend-hid.cpp:1205) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
[realsense2_camera_node-1]  30/01 23:18:37,668 WARNING [139936804103744] (backend-hid.cpp:1205) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.5.auto
[realsense2_camera_node-1]  30/01 23:18:37,668 WARNING [139936384665152] (backend-hid.h:48) write_fs_attribute with accel_3d-dev1 failed. The specified path /sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1.1/4-1.1:1.5/0003:8086:0B5C.003C/HID-SENSOR-200073.7.auto/iio:device1/trigger/current_trigger is not valid
[realsense2_camera_node-1]  30/01 23:18:37,668 ERROR [139936644707904] (librealsense-exception.h:52) Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1.1/4-1.1:1.5/0003:8086:0B5C.003C/HID-SENSOR-200073.7.auto/iio:device1/scan_elements/in_accel_y_en Last Error: Read-only file system
[realsense2_camera_node-1]  30/01 23:18:37,669 ERROR [139936644707904] (backend-hid.cpp:1005) Hid device is busy!
[realsense2_camera_node-1] [ERROR] [1706656717.669156742] [camera.camera]: ./src/rs_node_setup.cpp:370:An exception has been thrown: Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1.1/4-1.1:1.5/0003:8086:0B5C.003C/HID-SENSOR-200073.7.auto/iio:device1/scan_elements/in_accel_y_en Last Error: Read-only file system
[realsense2_camera_node-1] [ERROR] [1706656717.669198501] [camera.camera]: Error updating the sensors: Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1.1/4-1.1:1.5/0003:8086:0B5C.003C/HID-SENSOR-200073.7.auto/iio:device1/scan_elements/in_accel_y_en Last Error: Read-only file system
[realsense2_camera_node-1]  30/01 23:18:37,694 WARNING [139936804103744] (backend-hid.cpp:1205) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto
[realsense2_camera_node-1]  30/01 23:18:37,694 WARNING [139936804103744] (backend-hid.cpp:1205) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
[realsense2_camera_node-1]  30/01 23:18:37,694 WARNING [139936804103744] (backend-hid.cpp:1205) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.5.auto
[realsense2_camera_node-1] [INFO] [1706656717.695129269] [camera.camera]: Device with serial number 318122301758 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1706656717.695212644] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1.1/4-1.1:1.0/video4linux/video4 was found.
[realsense2_camera_node-1] [INFO] [1706656717.695230950] [camera.camera]: Device with name Intel RealSense D456 was found.
[realsense2_camera_node-1] [INFO] [1706656717.695577886] [camera.camera]: Device with port number 4-1.1 was found.
[realsense2_camera_node-1] [INFO] [1706656717.695596253] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1706656717.695807486] [camera.camera]: Stop Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1706656717.695832956] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [WARN] [1706656718.101862981] [camera.camera]: 
[realsense2_camera_node-1]  30/01 23:18:38,267 ERROR [139936795711040] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error
[realsense2_camera_node-1]  30/01 23:18:38,267 ERROR [139936795711040] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error
[realsense2_camera_node-1]  30/01 23:18:38,280 ERROR [139936636315200] (backend-v4l2.cpp:2771) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 20 error: No such device
[realsense2_camera_node-1]  30/01 23:18:38,280 ERROR [139936636315200] (backend-v4l2.cpp:2771) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 20 error: No such device
[realsense2_camera_node-1]  30/01 23:18:38,280 ERROR [139936636315200] (backend-v4l2.cpp:2771) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 20 error: No such device
```

MartyG-RealSense, 2024-01-31:
Hi @ashBabu  At https://github.com/IntelRealSense/realsense-ros/issues/2470 there was a case where a Docker user was able to access the IMU in librealsense 2.48.0 but not in 2.50.0 (the version that the Docker guide that you linked to was introduced in).

In a more recent case at https://github.com/IntelRealSense/librealsense/issues/11742 another user found that they could not access the IMU through Docker with a D435i but could if they used a RealSense T265 Tracking camera (a now retired RealSense model that had the same IMU component as D435i).  Their Dockerfile can be found here:

https://github.com/EnthusiasticRoboticist/DonkeyJet/blob/main/realsense.Dockerfile

So it appears that IMU remains inaccessible on Docker in the years since the guide that you linked to was published, unfortunately.

ashBabu, 2024-01-31:
Thanks @MartyG-RealSense. I am not an expert but does modifying some udev rules help?. In the file I linked, there are these lines `-v /dev:/dev 
    --device-cgroup-rule "c 81:* rmw" 
    --device-cgroup-rule "c 189:* rmw" ` passed to the docker container. When I checked `ls -l /dev`, the number 81 corresponds to a group `plugdev` or `video`. In another forum, I found `"c 10:* rmw"` would help in getting imu data but 10 corresponds to `root` or `kmv`.  Tried this but didnt work for me. So far the only thing that worked is to run docker with `privileged`. This gives access to imu data but it is not recommened. 

Not sure if the above is of any help but just a guess. Please feel free to close this if this is a dead end as I dont have T265 cam.

ashBabu, 2024-01-31:
I dont think `sudo` is going to help. When docker is installed, by default, all containers are run with `sudo docker run ...`. By adding docker to the groups by `sudo usermod -aG docker $USER` makes it run without the `sudo`

MartyG-RealSense, 2024-01-31:
In regard to your earlier comment about using `privileged `not being recommended, in RealSense Docker cases the use of privileged often **is** recommended in advice by RealSense users.

ashBabu, 2024-01-31:
ok. Thank you @MartyG-RealSense . Closing this

closed 2024-01-31 by ashBabu

rtarun1, 2025-09-25:
Hello @ashBabu ,

I am running my container in `privileged`, yet im not getting the imu topic. I only get the following permission denied error. 

```bash
ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true unite_imu_method:=1
```

```bash

[realsense2_camera_node-1]  25/09 13:06:39,030 WARNING [129422128576064] (ds-calib-parsers.cpp:36) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[realsense2_camera_node-1] [INFO] [1758785799.032344782] [camera.camera]: Starting Sensor: Motion Module
[realsense2_camera_node-1]  25/09 13:06:39,032 WARNING [129421876893248] (backend-hid.h:48) write_fs_attribute with gyro_3d-dev1 failed. The specified path /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/trigger/current_trigger is not valid
[realsense2_camera_node-1]  25/09 13:06:39,033 ERROR [129422128576064] (librealsense-exception.h:52) Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied
[realsense2_camera_node-1]  25/09 13:06:39,033 ERROR [129422128576064] (librealsense-exception.h:52) Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied
[realsense2_camera_node-1]  25/09 13:06:39,033 ERROR [129422128576064] (librealsense-exception.h:52) Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_y_en Last Error: Permission denied
[realsense2_camera_node-1]  25/09 13:06:39,033 ERROR [129422128576064] (librealsense-exception.h:52) Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_timestamp_en Last Error: Permission denied
[realsense2_camera_node-1]  25/09 13:06:39,033 ERROR [129422128576064] (librealsense-exception.h:52) Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_z_en Last Error: Permission denied
[realsense2_camera_node-1]  25/09 13:06:39,033 ERROR [129422128576064] (backend-hid.cpp:1082) Hid device is busy!
[realsense2_camera_node-1]  25/09 13:06:39,033 ERROR [129422128576064] (rs.cpp:256) [rs2_open_multiple( sensor:0x59f759791210, profiles:0x75b564070798, count:2 ) Backend] Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied
[realsense2_camera_node-1] [ERROR] [1758785799.033481304] [camera.camera]: /home/container_user/husky_ws/src/dependencies/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:511:An exception has been thrown: Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied
[realsense2_camera_node-1] [ERROR] [1758785799.033569815] [camera.camera]: /home/container_user/husky_ws/src/dependencies/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:393:An exception has been thrown: Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied
[realsense2_camera_node-1] Failed to start device: Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied
[realsense2_camera_node-1] [INFO] [1758785799.033827134] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1758785799.037049780] [camera.camera]: Starting Sensor: Motion Module
[realsense2_camera_node-1]  25/09 13:06:39,037 WARNING [129421650421312] (backend-hid.h:48) write_fs_attribute with gyro_3d-dev1 failed. The specified path /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/trigger/current_trigger is not valid
[realsense2_camera_node-1]  25/09 13:06:39,037 ERROR [129421910464064] (librealsense-exception.h:52) Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied
[realsense2_camera_node-1]  25/09 13:06:39,037 ERROR [129421910464064] (librealsense-exception.h:52) Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied
[realsense2_camera_node-1]  25/09 13:06:39,037 ERROR [129421910464064] (librealsense-exception.h:52) Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0014/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_y_en Last Error: Permission denied
[... cut ...] ed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0015/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied
[realsense2_camera_node-1] [ERROR] [1758785801.156112394] [camera.camera]: Error updating the sensors: Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B5C.0015/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied
```

ashBabu, 2025-09-25:
Looking at your logs, it mostly point to a `permission denied` error. Not sure but I am assuming that you are probably a non-root user who does not have access to `/dev`. 
Inside docker, try running `rs-enumerate-devices`. Towards the end of the log, you should see
```
   Color           |          YUYV        @ 60/30/6 Hz

Stream Profiles supported by Motion Module
 Supported modes:
    STREAM      FORMAT         FPS
    Accel       MOTION_XYZ32F  @ 250/63 Hz
    Gyro        MOTION_XYZ32F  @ 400/200 Hz
```
If this is obtained, then IMU should work inside docker.  May be worth checking `docker exec --user root -it <CONT-NAME> /bin/bash` and running `rs-enumerate-devices` again