realsenseai/librealsense #12625: aligned depth image and RGB image in L515

OneShz, 2024-01-31:
By using L515 ,I have already got the rgb images and depth images,but how to align them one by one ?

OneShz, 2024-01-31:
i have got the inner intrinsics {"fx": 599.6552734375, "fy": 599.66796875, "ppx": 325.10821533203125, "ppy": 235.8369598388672, "height": 480, "width": 640, "depth_scale": 0.0002500000118743628} like this, but when i use them to align the point cloud is not the same as i got by using the  code-(rosbag record -o  test1.bag  /camera/depth/color/points).

MartyG-RealSense, 2024-01-31:
Hi @OneShz  There is a pyrealsense2 / Open3D pointcloud script that uses depth-color alignment at https://github.com/isl-org/Open3D/issues/473#issuecomment-508139946 that you could compare to your own script.

OneShz, 2024-01-31:
> Hi @OneShz There is a pyrealsense2 / Open3D pointcloud script that uses depth-color alignment at [isl-org/Open3D#473 (comment)](https://github.com/isl-org/Open3D/issues/473#issuecomment-508139946) that you could compare to your own script.

Thank you for your reply I will try.

OneShz, 2024-02-01:
[image]
> Hi @OneShz There is a pyrealsense2 / Open3D pointcloud script that uses depth-color alignment at [isl-org/Open3D#473 (comment)](https://github.com/isl-org/Open3D/issues/473#issuecomment-508139946) that you could compare to your own script.

I see the comment but it not use,because I need a postpone 3d points,in the example it generates a instant 3d points so it must be connect by the machine,while i have gotten the rgb images and depth images.The following are the 3d I generated and the instant got.
[image]

[image]

MartyG-RealSense, 2024-02-01:
There are very few references available about creating pointclouds using RealSense and Open3D, so this will be a difficult issue to diagnose unfortunately.

Where did you obtain your intrinsic values for your L515 from, please?  Each camera has unique intrinsic values because of the manufacturing process at the factory, so it is important to retrieve the values from the exact camera that is being used rather than use values from another camera.

The intrinsic values below for depth and color at 640x480 are from one of my own L515.

[image]

[image]

MartyG-RealSense, 2024-02-16:
Case closed due to no further comments received.

closed 2024-02-16 by MartyG-RealSense

kingchou007, 2024-11-30:
Hi, @MartyG-RealSense 

I notice L515 rgb can support 1280x720, but the max depth is 1024x768. like rs.align(rs.stream.depth)？

How should I align rgb and depth during the data recording?

MartyG-RealSense, 2024-11-30:
Hi @kingchou007  The RealSense SDK's **align processing block** component automatically adjusts for different depth and RGB resolutions when rs.align is used.