realsenseai/librealsense #12626: Issues with building librealsense on Orin Nano (ARK)

ta-jetson, 2024-01-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Operating System & Version |   Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.10.104-tegra           | 
| Platform                                 |NVIDIA Orin nano 8gb dev kit on ARK |
| SDK Version                            |  2.48.0           | 
| python Version                            |  3.8.10           | 
| cmake Version                            |  3.27.4           | 
| Jetpack Version                            |  5.1.1           |
| L4T Version                            |  35.3.1           | 
 

### Issue Description
Hey @MartyG-RealSense,
I was building librealsense 2.48.0 on my Orin Nano (ARK), and faced an error during it. This procedure was from https://github.com/IntelRealSense/librealsense/issues/12377 and  https://github.com/IntelRealSense/librealsense/issues/10891.

This was my procedure:
Downloaded and unzipped librealsense-2.48.0 from librealsense-releases.

cd librealsense
cd librealsense && mkdir build && cd build
sudo apt-get -y install python3 python3-dev libssl-dev libxinerama-dev libxcursor-dev libcanberra-gtk-module libcanberra-gtk3-module

cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3.8 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true -DFORCE_RSUSB_BACKEND=ON

Error:
-- Build files have been written to: /home/tsalla/librealsense/build/external-projects/pybind11
[ 12%] Performing update step for 'pybind11'
[ 25%] No patch step for 'pybind11'
[ 37%] No configure step for 'pybind11'
[ 50%] No build step for 'pybind11'
[ 62%] No install step for 'pybind11'
[ 75%] Completed 'pybind11'
[100%] Built target pybind11
CMake Deprecation Warning at build/third-party/pybind11/CMakeLists.txt:8 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- pybind11 v2.6.2 
CMake Warning (dev) at build/third-party/pybind11/tools/FindPythonLibsNew.cmake:98 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  build/third-party/pybind11/tools/pybind11Tools.cmake:45 (find_package)
  build/third-party/pybind11/tools/pybind11Common.cmake:201 (include)
  build/third-party/pybind11/CMakeLists.txt:188 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /usr/local/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python (missing: Python_INCLUDE_DIRS Development
  Development.Module Development.Embed) (found version "2.7.18")
Call Stack (most recent call first):
  /usr/local/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.27/Modules/FindPython.cmake:616 (find_package_handle_standard_args)
  wrappers/python/CMakeLists.txt:62 (find_package)
  
  
  i have looked at https://github.com/IntelRealSense/librealsense/issues/10891 and https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049, which seemed to solve the same problem for the rest.
  I even updated my cmake from 3.16 to 3.27 to see if this error is sorted. Tried building librealsense 2.50.0 as well. 
  I tried to change -DPYTHON_EXECUTABLE=/usr/bin/python3.8 to -DPYTHON_EXECUTABLE=/usr/bin/python3.
  Only change in error being :
  
  -- Build files have been written to: /home/tsalla/librealsense/build/external-projects/pybind11
[ 12%] Performing update step for 'pybind11'
[ 25%] No patch step for 'pybind11'
[ 37%] No configure step for 'pybind11'
[ 50%] No build step for 'pybind11'
[ 62%] No install step for 'pybind11'
[ 75%] Completed 'pybind11'
[100%] Built target pybind11
CMake Deprecation Warning at build/third-party/pybind11/CMakeLists.txt:8 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- pybind11 v2.6.2 
CMake Warning (dev) at build/third-party/pybind11/tools/FindPythonLibsNew.cmake:98 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  build/third-party/pybind11/tools/pybind11Tools.cmake:45 (find_package)
  build/third-party/pybind11/tools/pybind11Common.cmake:201 (include)
  build/third-party/pybind11/CMakeLists.txt:188 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10")     *This is the only change*
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython3.8.so
CMake Error at /usr/local/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python (missing: Python_INCLUDE_DIRS Development
  Development.Module Development.Embed) (found version "2.7.18")
Call Stack (most recent call first):
  /usr/local/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.27/Modules/FindPython.cmake:616 (find_package_handle_standard_args)
  wrappers/python/CMakeLists.txt:62 (find_package)

  
  
  
  
  
  No luck with any step. 
 Could you please suggest on how to proceed with this error?

MartyG-RealSense, 2024-01-31:
Hi @ta-jetson  A RealSense Jetson user at https://github.com/IntelRealSense/librealsense/issues/11634 experienced these errors and resolved it with a workaround of making an edit to the Python wrapper's CMakeLists.txt file.  It should not be necessary to do this though in order to achieve a working installation.

A popular guide to installing librealsense and the Python wrapper on Jetson boards from source code can be found at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

**********

In regard to your apt-get instruction:

`sudo apt-get -y install python3 python3-dev libssl-dev libxinerama-dev libxcursor-dev libcanberra-gtk-module libcanberra-gtk3-module`

The official Jetson source code installation instructions at the link below suggest using the following apt-get command:

`sudo apt-get install git libssl-dev libusb-1.0-0-dev libudev-dev pkg-config libgtk-3-dev -y`

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

Of these recommended commands, only `libssl-dev` is included in your own apt-get instruction.

ta-jetson, 2024-02-01:
hey @MartyG-RealSense ,

I tried the new installation instructions provided by you, but the error persists. I had already referred to https://github.com/IntelRealSense/librealsense/issues/11634 and https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 before opening my issue. 

----------------------------------------------------------------------------------------

Regarding the CMakelists.txt, I added the following
set(PYTHON_EXECUTABLE "/usr/bin/python3.8")
set(PYTHON_INCLUDE_DIR "/usr/include/python3.8")

Error:
-- Build files have been written to: /home/tsalla/librealsense/build/external-projects/pybind11
[ 12%] Performing update step for 'pybind11'
[ 25%] No patch step for 'pybind11'
[ 37%] No configure step for 'pybind11'
[ 50%] No build step for 'pybind11'
[ 62%] No install step for 'pybind11'
[ 75%] Completed 'pybind11'
[100%] Built target pybind11
CMake Deprecation Warning at build/third-party/pybind11/CMakeLists.txt:8 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- pybind11 v2.6.2 
CMake Warning (dev) at build/third-party/pybind11/tools/FindPythonLibsNew.cmake:98 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  build/third-party/pybind11/tools/pybind11Tools.cmake:45 (find_package)
  build/third-party/pybind11/tools/pybind11Common.cmake:201 (include)
  build/third-party/pybind11/CMakeLists.txt:188 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/bin/python3.8 (found version "3.8.10") 
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython3.8.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
CMake Error at /usr/local/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python (missing: Python_INCLUDE_DIRS Development
  Development.Module Development.Embed) (found version "2.7.18")
Call Stack (most recent call first):
  /usr/local/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.27/Modules/FindPython.cmake:616 (find_package_handle_standard_args)
  wrappers/python/CMakeLists.txt:66 (find_package)

-------------------------------------------------------------------------------------

A user had mentioned to delete the development directory ([Development directory from the cmake find python](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/CMakeLists.txt#L63)). That is something I haven't tried yet, cause I do not know where this directory is there. 
Could you suggest what other changes in CMakelists.txt can be made?

MartyG-RealSense, 2024-02-01:
I will ask the RealSense user who posted about the development directory workaround.

@bhomaidan1990  Would it be possible to advise @ta-jetson please about how you removed the `development `directory in your workaround at https://github.com/IntelRealSense/librealsense/issues/11634 please?  Thanks!

ta-jetson, 2024-02-01:
Hey @MartyG-RealSense ,

Is there any work around to this? What about any changes to the cmakelist that'll solve the problem?

MartyG-RealSense, 2024-02-01:
@ta-jetson  There is not a clear reason why you are having these build difficulties.

It may be easier if you are able to use the latest librealsense version (2.54.2) and then install the pyrealsense2 wrapper separately after librealsense has been successfully installed.

First, install librealsense with the JetsonHacks build script **installLibrealsense.sh** at the link below.

https://github.com/JetsonHacksNano/installLibrealsense

Then build the pyrealsense2 wrapper from packages with the command **pip install pyrealsense2**

The pip install method can work with Python 3.8 on computing devices with the Arm architecture, as Python 3.7, 3.8 and 3.9 are currently supported by pip install pyrealsense2 on Arm.

ta-jetson, 2024-02-01:
Hey @MartyG-RealSense ,
I specifically need librealsense 2.50.0 and below. 
My setup has both T265 and D435i. 
One thing I noticed is i can compile it without python bindings successfully. But I need python bindings for my algorithms.
I tried to
- install python3-pybind11 separately. 
- change c++11 to c++14

I am actively trying to troubleshoot this. Do you know any areas or references to look into that could most likely be the root of this issue?
Any suggestion would be helpful

MartyG-RealSense, 2024-02-01:
In this situation and the requirement for a librealsense version no newer than 2.50.0, the only option that may be available is to compile librealsense first from source code without the Python wrapper to achieve a successful build and then compile the Python wrapper **separately** from source code afterwards using the instructions at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source

ta-jetson, 2024-02-02:
Hey @MartyG-RealSense ,
An update on the situation. I can build and use librealsense (realsense-viewer) without the python bindings. If I compile the wrapper after successful build, I get the same error.

These are the additional changes I tried :

find_package(Python${PYTHON_VERSION} COMPONENTS Development.Module)
find_package(Python${PYTHON_VERSION}  REQUIRED COMPONENTS Interpreter Development)
find_package(Python3 REQUIRED COMPONENTS Interpreter Development)
set(PYTHON_EXECUTABLE "/usr/bin/python3.8")
set(PYTHON_INCLUDE_DIR "/usr/include/python3.8")
set(PYTHON_LIBRARY "/usr/lib/aarch64-linux-gnu/libpython3.8.so") 

And changed Python REQUIRED COMPONENTS to Python3 REQUIRED COMPONENTS.

Error still persists. I am looking into the python development.module and will look into this error further. Do you have any suggestions or references to look into that could most likely be the root of this issue?

ta-jetson, 2024-02-02:
Hey @MartyG-RealSense ,

I solved the issue. 

I'll expand on my solution.

1. Download the librealsense-2.50.0 from the librealsense releases and unzip it.
2. In ~/librealsense-2.50.0/wrappers.python/CMakelists.txt, Remove/comment line-63 or the line:
find_package(Python3 REQUIRED COMPONENTS Interpreter Development)
3. Go back to build folder.
4. Run the following commands
--------------------------------------------------------------------

```
$ cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3.8 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true -DFORCE_RSUSB_BACKEND=ON

$ sudo make uninstall && make clean

$ make -j6  

$sudo make install

```

Note: Mine was -j6 run with the lower number if you have less cores. You can check by using the command "nproc" on terminal

-------------------------------------------------------------------

In .bashrc :

```
PYTHONPATH="/usr/local/lib:/usr/local/lib/python3.8/pyrealsense2:$PYTHONPATH"
export PYTHONPATH

```

--------------------------------------------------------------------------------------

To get pyrealsense2 to run everywhere 

```
$ sudo mkdir /usr/local/lib/python3.8/pyrealsense2

$ sudo ln -s ~/librealsense-2.50.0/build/wrappers/python/pybackend2.cpython-38-aarch64-linux-gnu.so /usr/local/lib/python3.8/pyrealsense2/

$ sudo ln -s ~/librealsense-2.50.0/build/wrappers/python/pyrealsense2.cpython-38-aarch64-linux-gnu.so /usr/local/lib/python3.8/pyrealsense2/

$ source ~/.bashrc

```

Thanks again @MartyG-RealSense.

closed 2024-02-02 by ta-jetson

MartyG-RealSense, 2024-02-02:
It's great to hear that you achieved a solution.  Thanks so much for sharing the details of it!