realsenseai/librealsense #12627: After suggestion of the firmware update d455 device crashed. doesn't detect by the system

MetehanGokmen, 2024-01-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu 20.04} | 
| Kernel Version (Linux Only)    |  (5.15.0-92-generic)                                         | 
| Platform                                 | PC |
| SDK Version                            |  {2.54.2}                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
I didn't have a stable connection on my ubuntu device so I used in Windows. When I try to use it, a pop up showed me. It says We suggest a firmware update. Then I updated it. After that. my laptop see a device but realsense viewer van't see the my device. I tried both ubuntu and windows options but it didn't work. I all almost all issues about it and still didn't get any solution. Last year we had the same probe and I sent the camera to service after the return it still have same issus about like
"frames didn't arrive in 5 seconds"
but it works sometime. 
Now it is not detected by realsense-viewer and update tools. I am getting this error.

rs-fw-update -l

Connected devices:
 31/01 23:38:26,391 WARNING [140323323875712] (rs.cpp:312) null pointer passed for argument "device"
RealSense error calling rs2_create_device(info_list:0x55b70c606a60, index:0):
    hwmon command 0x10( 0 0 0 0 ) failed (response -844431340= unknown)

lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8086:0ad6 Intel Corp. Intel(R) RealSense(TM) Invalid SKU

is there any new solution to solve this problem?
I tried all possible solutions.

MartyG-RealSense, 2024-02-01:
Hi @MetehanGokmen  I located your previous camera return case at https://github.com/IntelRealSense/librealsense/issues/11630

I will speak to my Intel RealSense colleagues about your new problem.  Thanks very much for your patience.

MetehanGokmen, 2024-02-01:
I am in  Turkiye. Thanks for reply.

MetehanGokmen, 2024-02-01:
Hey, is there any solution? @MartyG-RealSense

MartyG-RealSense, 2024-02-01:
I am discussing your issue with my colleagues and will let you know as soon as I have information to share with you.  Thanks again.

MartyG-RealSense, 2024-02-02:
I have spoken to my colleagues.  Please send me an email so that I can discuss the next step with you to resolve your issue.

martyx.grover@intel.com

MetehanGokmen, 2024-02-02:
I sent an email about the last situation and how it happened in detail. @MartyG-RealSense

MartyG-RealSense, 2024-05-18:
Hi @MetehanGokmen  Has this case been resolved to your satisfaction, please?  Thanks!

closed 2024-06-03 by MartyG-RealSense