realsenseai/librealsense #12631: [L515] What is the reference frame of the 3D point got from rs2_deproject_pixel_to_point()?

JeongJunLee, 2024-02-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.8.1 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    | N/A                                     | 
| Platform                                 | PC  |
| SDK Version                            | Intel.RealSense.SDK-WIN10-2.54.2.5684                       | 
| Language                            | C/C++                        | 
| Segment			|  Robot                        | 

I could successfully get the X,Y,Z coordinate of 3D points by following functions:
------
float pp[2];
float ppp[3]
auto distance = depth_aligned.get_distance( (int) pp[0],  (int) pp[1]);
rs2_deproject_pixel_to_point(ppp, &depth_intrin, pp, distance);
-----

Now I am wondering what is the reference frame of the above coordinates (ppp[3]). 
Is it the "depth camera reference frame" whose origin is lens center of depth camera? 
or, is it related to "IR Transmitter (for X,Y)" and  "Front cover glass (for Z)"? 

From the document "Intel_RealSense_LiDAR_L515_Datasheet_Rev003.pdf", 
it looks like that it is "IR Transmitter (for X,Y)" and  "Front cover glass (for Z)". 

But I want to confirm it. 
And if so, how can I change the reference frame from "IR Transmitter and Front Cover Glass" reference frame to "depth camera" reference frame? 
Just add 4.5mm in Z coordinate? Is it all? 

Thanks in advance.

MartyG-RealSense, 2024-02-02:
Hi @JeongJunLee  On the L515 camera model the '0' depth start point of depth measurement is the front glass.  

The depth **origin** coordinates are at the XY center of the IR transmitter.

In the majority of cases the depth measurement from the front glass, provided by **get_distance()**, will be the value that is used.

If you need the 'ground truth' measurement from behind the front glass instead of the measurement from the glass though then you can add a minus-value of **-4.5 mm** to the depth values.

JeongJunLee, 2024-02-02:
Hi MartyG, 

Thank you for the answer. 
You mean that the output of below function rs2_deproject_pixel_to_point(), ppp[] is 'IR Transmitter + Front glass' based coordinate, right? 
//////
float pp[2];
float ppp[3]
auto distance = depth_aligned.get_distance( (int) pp[0], (int) pp[1]);
rs2_deproject_pixel_to_point(ppp, &depth_intrin, pp, distance);
//////

Now then, what I want to know is 
"how I can change the coordinate in 'IR Transmitter + Front glass' reference frame to that in 'depth camera' reference frame". 
As I said before what I want to know is 'depth camera' based coordinate. 

Please, can you let me know how I can do it? 

p.s. 
It looks like another function below 
    rs2_project_point_to_pixel(pp, &depth_intrin, ppp);
changes 'IR Transmitter + Front glass' based coordinate (ppp) to "depth camera" based coordinate, 
and computes from the "depth camera" based coordinate  the 2D coordinate on depth image (pp).

Thanks in advance.

JeongJunLee, 2024-02-07:
Hello MartyG, 
It looks like that 3D positions obtained by rs2_deproject_pixel_to_point() is "depth camera based coordinate". 
Anyway, thank you for your time and support. 
appreciate~~

closed 2024-02-07 by JeongJunLee