realsenseai/librealsense #12632: Unable to Change Gain via Python

ShankarKandasamy, 2024-02-02:
When I try to set the gain on my L515 using the gain method in the pyrealsense2.option class; I get the following error:
"RuntimeError: hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
Very frustrating as I've tried changing USB ports, computers, firmware is the latest available.
I'm also able to adjust the gain using the RS Viewer without a problem.
Changing the gain is important to my application, so I need to figure this out.

Below is my current code:
"""
  import pyrealsense2 as rs
  import numpy as np
  import cv2
  
  # Configure depth and color streams
  pipeline = rs.pipeline()
  config = rs.config()
  config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
  config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
  
  # Start the pipeline
  pipeline.start(config)
  
  try:
      # Get the active profile and select the color sensor
      profile = pipeline.get_active_profile()
      color_sensor = profile.get_device().first_color_sensor()
  
      # Check if the gain option is supported
      if color_sensor.supports(rs.option.gain):
          print("Gain adjustment is supported.")
  
          # Get the supported range of the gain
          gain_range = color_sensor.get_option_range(rs.option.gain)
          print(f"Supported gain range: {gain_range.min} to {gain_range.max}")
  
          # Set a gain value within the supported range
          desired_gain_value = 16.0  # Ensure this value is within the supported range
          if gain_range.min <= desired_gain_value <= gain_range.max:
              color_sensor.set_option(rs.option.gain, desired_gain_value)
              print(f"Gain successfully set to {desired_gain_value}.")
          else:
              print(
                  f"Desired gain value {desired_gain_value} is out of the supported range."
              )
      else:
          print("Gain adjustment is not supported.")
  
      while True:
          # Wait for a coherent pair of frames: depth and color
          frames = pipeline.wait_for_frames()
          depth_frame = frames.get_depth_frame()
          color_frame = frames.get_color_frame()
          if not depth_frame or not color_frame:
              continue
  
          # Convert images to numpy arrays
          depth_image = np.asanyarray(depth_frame.get_data())
          color_image = np.asanyarray(color_frame.get_data())
  
          # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
          depth_colormap = cv2.applyColorMap(
              cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET
          )
  
          # Stack both images horizontally
          images = np.hstack((color_image, depth_colormap))
  
          # Show images
          cv2.imshow("RealSense", images)
  
          # Break loop with 'q'
          if cv2.waitKey(1) & 0xFF == ord("q"):
              break
  finally:
      # Stop streaming
      pipeline.stop()
      cv2.destroyAllWindows()
    """

MartyG-RealSense, 2024-02-02:
Hi @ShankarKandasamy  I found that I encountered this error with the RGB color gain in the RealSense Viewer too with two different L515.  It worked if I disabled RGB Auto Exposure before changing the setting.  So it may be worth disabling Auto Exposure on the color sensor in your script before setting Gain to see if it resolves your issue.

ShankarKandasamy, 2024-02-03:
Hi @MartyG-RealSense, this worked! 
Thanks a million, I wouldn't have thought to try that. 
When we change the 'gain', I guess the SDK just adjusts the exposure time to suite?

MartyG-RealSense, 2024-02-03:
You are very welcome!

Changing RGB gain does not change exposure.  Like an exposure change, changing gain does alter the brightness of the image.  Increasing gain above the default value adds more noise to the image.

ShankarKandasamy, 2024-02-05:
Thanks, @MartyG-RealSense. You can consider this matter closed and thanks again for the prompt solution. Cheers!

MartyG-RealSense, 2024-02-05:
You are very welcome, @ShankarKandasamy - thanks very much for the update!  As you suggested, I will close this issue.  Thanks again!

closed 2024-02-05 by MartyG-RealSense