realsenseai/librealsense #12633: Unhandled exception escaping from a worker thread!

Mosig33, 2024-02-04:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2024-02-04:
Hi @Mosig33  Please first check whether permission to access the camera hardware and permission for applications to access the camera are both set to 'On' in the Windows camera privacy settings.  This can be done using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

Mosig33, 2024-02-04:
Hi, thanks for the feedback. Just to add, everything you suggested is already done, and the crash happened after i was out of the Camera Video for like 10mins... When i came back, i saw he crash message.

MartyG-RealSense, 2024-02-05:
Which firmware driver version is your camera using, please?  You can find the firmware version in the RealSense Viewer by clicking on the 'Info' button at the top of the Viewer's options side-panel.  Firmware **5.15.1.0** should be used with 2.54.2.

closed 2024-02-20 by MartyG-RealSense