realsenseai/librealsense #12634: Tf data error

hirealsense, 2024-02-06:
| Camera Model                       | { D455 } | 
| Operating System & Version |   {Linux (Ubuntu 20.04)  | 
| SDK Version                            |  {2.54.2 }    | 

I tried opensource_tracking.launch with realsense D455
When RVIZ is running and I move the camera even a little bit, the TF DATA moves quickly and disappears
I thought it might be a problem with the imu, but imu calibration didn't work

https://github.com/IntelRealSense/librealsense/assets/159091346/fbe06768-3c39-41f1-a94b-dc9dfe5ae1c4

MartyG-RealSense, 2024-02-06:
Hi @hirealsense  Are your little movements of the camera fast movements?  Intel's D435i SLAM guide which makes use of opensource_tracking.launch states that "the built-in IMU can only keep track for a very short time.  Moving or turning too quickly will break the sequence of successful point cloud matches and will result in the system losing track".

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

You could try using 'copy' for the unite_imu_method parameter instead of 'linear_interpolation', as IMU data may be more stable in Copy mode.

`roslaunch realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true unite_imu_method:=copy`

hirealsense, 2024-02-07:
thank you @MartyG-RealSense I tried what you said but the problem persists. And i tried adjusting the imu threshold value in the rtabmap.ini file, but it didn't work

MartyG-RealSense, 2024-02-07:
What happens if you set the frequency of the IMU streams to 200 Accel and 400 Gyro by adding the commands below onto the end of your opensource_tracking.launch instruction?

`accel_fps:=200 gyro_fps:=400`

hirealsense, 2024-02-08:
It takes longer than before to lose TFdata, but it still happens. Also, I have a new issue: TFdata moves as soon as I start opensource_tracking.launch

I tried initialization and reinstalling my Ubuntu, but the problem is still there, so I think it might be a device issue, and I'd like to know what you think about it.

MartyG-RealSense, 2024-02-08:
My recollection is that IMU tracking can also be worse if the camera is observing a scene without many "interesting" details in it.  More information about this can be found at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/8338902895379/comments/11237875710355

closed 2025-01-06 by MartyG-RealSense