realsenseai/librealsense #12638: Get the 3D coordinate of a point under camera frame

yijionglin, 2024-02-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435| 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |   Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  6.2.0-34                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |   2.54.2.5684                           | 
| Language                            |  {python                           | 
| Segment			|  Robot                          | 

### Issue Description
Hi there,

I am trying to get the point cloud of an object that is segmented from an image. Things I am able to get now are:
1. The camera-to-gripper transform matrix (through hand-eye calibration).
2. The pixel alignment of color frames and depth frames (through the code provided [here](https://github.com/IntelRealSense/librealsense/issues/5603))
3. The distance from a desired point (specified by pixel index in a depth image) to the camera using depth_frame.get_distance()

My question is, how to get the coordinate under the camera frame of a point specified in a color image? For example, is there any way that I could get the 3D coordinate of the blue point under the camera frame, so that I could later direct my robot to reach the blue point through camera-to-gripper transformation?

[image]

yijionglin, 2024-02-06:
I know that we can obtain the 3D coordinates of the point cloud using rs.pointcloud.get_vertices. 

However, I have a few questions: 
1. How do we determine the mapping between pixels in the depth or color frame and the corresponding indices in the 3D coordinate space, given the frame sizes e.g., (640, 480) and the coordinate array e.g. (307200, 3))? 
2. Are these vertices coordinated with the depth frame? 
3. If they are, does this imply that we need to perform additional transformations from the depth frame to the color frame, considering that my camera-to-gripper transform is calibrated with the color images?

yijionglin, 2024-02-06:
Update:

I just tried to get the 3D coordinate from the using this mapping:
`    ind = depth_point_[1]*640 + depth_point_[0]
    print(vertices[ind])
`
And the z coordinate seems the same as the one from get_distance.

And then I tried to compare the 3D coordinate computed from the checkerboard-to-camera and the one from `pts.get_vertices()[ind]`. 

Specifically, the first image was used for calculating the 3D coordinate (of the first red point) under the camera frame using solvePnP, which is 
[-109.72903171, -49.13718998 , 303.12990131] mm.
[image]

And the second image (technically, color and depth images were used) was used for calculating the 3D coordinate of the blue point (which was aligned with the previous red point position) through the following code, and the result is : [-0.12397843, -0.04965379,  0.30100003 ]. 

The results along y and z axes are pretty the same, except there is an offset of around 15mm along x axis which I guess this is the distance between the IR cam and the RGB cam. This is nice, but I am wondering if there is any (standard) way to do the whole thing more elegantly.

[image]

`color_point =  np.asarray([100, 150])
color_point = np.round(color_point).astype(int)

while True:
    frames = pipeline.wait_for_frames()
    color_frame = frames.get_color_frame()
    depth_frame = frames.get_depth_frame()

    color_image = np.asanyarray(color_frame.get_data())
    depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())
    # set_trace()
    # for color_point in color_points:    
    depth_point_ = rs.rs2_project_color_pixel_to_depth_pixel(
                depth_frame.get_data(), depth_scale,
                depth_min, depth_max,
                depth_intrin, color_intrin,  color_to_depth_extrin,depth_to_color_extrin, color_point)

    # Draw points
    color = (0, 0, 255)  # Red color in BGR format (Blue, Green, Red)
    thickness = 10  # You can adjust the thickness as needed

    cv2.circle(color_image, color_point, 1, color, thickness)
    depth_point_ = np.round(depth_point_).astype(int)
    cv2.circle(depth_colormap, depth_point_, 1, color, thickness)

    images = np.hstack((cv2.cvtColor(color_image, cv2.COLOR_BGR2RGB), depth_colormap))
    cv2.imshow("win", images)
    key = cv2.waitKey(1)
    # print(depth_point_)

# set_trace()
    
    # Point cloud
    # set_trace()
    if depth_point_[0]>0:
        print(depth_frame.get_distance(depth_point_[0],depth_point_[1]))
    pts = pc.calculate(depth_frame)
    v = pts.get_vertices()
    vertices = np.asanyarray(v).view(np.float32).reshape(-1, 3)
    # set_trace()
    ind = depth_point_[1]*640 + depth_point_[0]
    print(vertices[ind])`

MartyG-RealSense, 2024-02-07:
Hi @yijionglin   Thanks so much for your very detailed updates!

**get_vertices** is more accurate than using **align_to** and **rs2_deproject_pixel_to_point** to obtain 3D coordinates, so you are already using the optimal coordinate retrieval method.

yijionglin, 2024-02-07:
Hi @MartyG-RealSense ,

Thanks for your quick reply and the confirmation (which definitely makes me feel safer)!

Still have two quick questions if you don't mind:
1. Is there any standard method to get the 3D coordinate of a specified pixel in the color image?
2. Is there any standard method to get a transform matrix that maps the 3D coordinates of points under the depth frame to the color frame? (Currently I do it by subtracting the offset along the x-axis (15mm) but it makes me feel unsafe...)

Thanks again for your great effort in maintaining the community here!

Best,
Yijiong

MartyG-RealSense, 2024-02-08:
You are very welcome.  Thanks for your kind words!

1.  If you need a 3D XYZ coordinate then there are two main approaches, **vertices** (points.get_vertices) or **deprojection** (rs2_deproject_pixel_to_point).  Usually with deprojection, depth-color alignment has to be first performed and the alignment can introduce inaccuracy, so vertices are a bit more accurate.

2.  Typically, a **map_to** instruction is placed before pc.calculate in order to map depth and color together.  for example:

```
pc.map_to(color);
pts = pc.calculate(depth_frame)
```

closed 2024-02-08 by yijionglin

yijionglin, 2024-02-08:
> You are very welcome. Thanks for your kind words!
> 
> 1. If you need a 3D XYZ coordinate then there are two main approaches, **vertices** (points.get_vertices) or **deprojection** (rs2_deproject_pixel_to_point).  Usually with deprojection, depth-color alignment has to be first performed and the alignment can introduce inaccuracy, so vertices are a bit more accurate.
> 2. Typically, a **map_to** instruction is placed before pc.calculate in order to map depth and color together.  For example:
> 
> ```
> pc.map_to(color_frame);
> pts = pc.calculate(depth_frame)
> ```

Hi @MartyG-RealSense ,

Thanks for you reply! However when I tried the code you provided: placed the pc.map_to function before pc.calculate, it still returned the same 3D coordinates which were under the depth frame, not color frame.

Here is the snippet of my code:
```
    color_point =  np.asarray([100, 150])
    depth_point_ = rs.rs2_project_color_pixel_to_depth_pixel(
                depth_frame.get_data(), depth_scale,
                depth_min, depth_max,
                depth_intrin, color_intrin,  color_to_depth_extrin,depth_to_color_extrin, color_point)

    pc.map_to(color_frame)
    pts = pc.calculate(depth_frame)
    v = pts.get_vertices()
    vertices = np.asanyarray(v).view(np.float32).reshape(-1, 3)
    ind = depth_point_[1]*640 + depth_point_[0]
    print(vertices[ind])
```

MartyG-RealSense, 2024-02-08:
As well as retrieving the depth coordinates you can also obtain **texture coordinates** from the color, as described at https://github.com/IntelRealSense/librealsense/issues/6339