realsenseai/librealsense #12641: Can't able to build pyrealsense2 from source

Muthukumar4796, 2024-02-08:
| Camera Model                       | D455 | 
| Operating System & Version | Ubunttu 22.04  | 
| Language                            | python | 

I'm trying to install Pyrealsense 2 by building it from source. However, during the build process, it keeps showing an error.

I have attached the error for your reference.[build_error.txt](https://github.com/IntelRealSense/librealsense/files/14204975/build_error.txt)

MartyG-RealSense, 2024-02-08:
Hi @Muthukumar4796  Are you able to install the librealsense SDK without the pyrealsense2 wrapper?  If so then you could afterwards install the wrapper separately from packages instead of source code with the command **pip install pyrealsense2**

At the time of writing this, the pip install method of installing the wrapper supports Python 3.7 to 3.11 on computers / computing devices with x86 / x64 architecture (such as laptop and desktop PCs) and Python 3.7 to 3,9 on Arm architectured devices (such as Nvidia Jetson and Raspberry Pi).

Muthukumar4796, 2024-02-08:
My main goal is to fully build the pyrealsense2 library from source with CUDA on my Ubuntu 22.04 PC, and ensure that it only uses the GPU and not the CPU. If you have any ideas on how to accomplish this, please provide step-by-step instructions.

Muthukumar4796, 2024-02-08:
I have a laptop with a built-in Nvidia-1650 graphics card that supports CUDA version 12.2. It runs Ubuntu 22.04. However, you mentioned that **Nvidia-equipped PCs not work for building pyrealsense2**. Therefore, **I am unable to build pyrealsense2 for my PC**. Could you suggest an alternative method for me to do this?

MartyG-RealSense, 2024-02-08:
PCs with an Nvidia GPU have difficulty enabling librealsense's CUDA support but CUDA is not required for pyrealsense2 and you can build the wrapper on PC without enabling it.

Muthukumar4796, 2024-02-08:
Yes but it consumes CPU, and I have another process that requires CPU power.Thats why I tried to build with GPU.

MartyG-RealSense, 2024-02-09:
The librealsense SDK has an alternative to CUDA for offloading work onto the GPU called **GLSL processing blocks**, and it is 'vendor neutral' meaning that it can work with any GPU brand and not just Nvidia.  However, it is meant for use with C++ and so may not be helpful to you if your project requires Python.  Please see https://github.com/IntelRealSense/librealsense/pull/3654 for further details about GLSL.

There are not other options than CUDA and GLSL for offloading RealSense processing to the GPU, unfortunately.

closed 2025-01-06 by MartyG-RealSense