realsenseai/librealsense #12643: Kinova Gen3 - Networked Realsense - Can't access Depth Stream

saulcarvalho, 2024-02-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D410 } | 
| Firmware Version                   | 2.53.1 | 
| Operating System & Version | Linux (Ubuntu 20.04 LTS) | 
| Kernel Version (Linux Only)   |  5.15.0-92-generic | 
| Platform                                 | PC |
| SDK Version                           | 2.53.1       | 
| Language                            |  Python                         | 
| Segment			|  Robot                      | 

### Issue Description

I am using a Kinova Gen3 and I'm trying to use the D410 camera (I suppose it is this version). Since the camera is integrated into the robot, I can only access it by network, in this case I'm using a local network connection through Ethernet. I should mention that I am using a Linux Virtual Machine (as said above) on a Windows PC. The camera isn't detected in neither of them.

Now to the problem itself, when I use MATLAB (in Windows) I can access both the color and depth streams with the respective Vision IMAQ adapter, but when using python I can only see the color stream, by accessing the RTSP stream in (rtsp://192.168.1.10/color). Ideally I want to access the depth stream in RTSP (rtsp://192.168.1.10/depth) with Python, but for some reason it never works, as if the RTSP endpoint isn't active at all.

I have used RealSense SDK version 2.54.1, but seems network mode is disabled, so I successfully downgraded it to 2.53.1. There I can add a Device IP, to which I tried the Kinova Gen3 default IP, which is working for the color stream (192.168.1.10), but I get the error [getStreams] error: client time out - 3.

I am obviously missing something, I tried applying the solution of other similar Issues here, but to no avail. I thought that updating the firmware of the camera would help, but since I can't even detect it in neither of the OS's, I don't know what to do.

In summary, all I want to do is access the depth stream in Python, since the color one is already working.

MartyG-RealSense, 2024-02-08:
Hi @saulcarvalho  Accessing the depth stream of a D410 on a Kinova Gen3 with Python code is discussed at https://github.com/IntelRealSense/librealsense/issues/11993

saulcarvalho, 2024-02-08:
Hello @MartyG-RealSense,

For https://github.com/IntelRealSense/librealsense/issues/11993:
- Tried the code that [gustn6591](https://github.com/gustn6591) posted, didn't work;
- Tried the Gstreamer solutions you posted, didn't work;

When I mean it doesn't work, is that I am not even able to open the depth stream, I can't detect the camera in any OS whatsoever.  RealSense Viewer doesn't detect anythin. Device manager doesn't detect anything. I can only see color stream, nothing else. If I try to use the pyrealsense2 library I can't do anything either, because it depends on the connection to the camera, and since I can't detect it, it doesn't work.

The only solution working so far is MATLAB's solution, but for my application it won't work: https://www.mathworks.com/help/supportpkg/robotmanipulator/ug/generate-colorized-point-cloud-gen3.html

I already followed other similar posts, but once again, they all end up depending on pyrealsense2, so no luck.

MartyG-RealSense, 2024-02-09:
I note that you tried to access the D410's depth stream through a network IP.    Were you using the RealSense Viewer tool to access this network connection please?

In order to use an IP to access the camera, the SDK needs to be built from source code with CMake with the flag **-DBUILD_NETWORK_DEVICE=ON** included in the CMake build instruction, and then the camera needs to be connected to with **ssh**.  Have you enabled networking using the guide below, please?

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

Having networking support enabled in the SDK is also necessary for accessing networking from a RealSense program script.

The RealSense D410 camera module is not equipped with a color sensor, only depth and infrared capability, so Kinova Gen3 provides its own separate color sensor component as shown in a diagram at https://github.com/IntelRealSense/librealsense/issues/12122#issuecomment-1688199073

MartyG-RealSense, 2024-02-23:
Case closed due to no further comments received.

closed 2024-02-23 by MartyG-RealSense

saulcarvalho, 2024-09-17:
@MartyG-RealSense Sorry for my lack of response at the time, but I eventually started working on other things. I recently got into that specific problem again and I found a workaround without using what I was looking at the time.

# Solution

System: Windows 11
Python Version:  3.11

1. Download and install MINGW
2. Download and install Gstreamer with all plugins
3. To be able to access streams with python, everything needs to be built with MSYS2, because pyobject could not be compiled by default Python compiler and required mingw64. MSYS2 was used:
        - Install PyGobject:
                - Go to https://www.msys2.org/ and download the x86_64 installer
                - Follow the instructions on the page for setting up the basic environment
                - Run C:\msys64\ucrt64.exe - a terminal window should pop up
                - Execute pacman -Suy
                - Execute pacman -S mingw-w64-ucrt-x86_64-gtk4 mingw-w64-ucrt-x86_64-python3 mingw-w64-ucrt-x86_64-python3-gobject
                - To test that GTK is working you can run gtk4-demo
                - Add C:\msys64\mingw64\bin as a python interpreter (e.g. my IDE was VS Code)
        - Install Gstreamer with MSYS2: 
                - pacman -S mingw-w64-x86_64-gst-pythonPP
                - pacman -S mingw-w64-x86_64-gst-plugins-good
                - pacman -S mingw-w64-x86_64-gst-plugins-bad 
                - pacman -S mingw-w64-x86_64-gst-plugins-ugly
                - pacman -S mingw-w64-x86_64-gst-libav
        - Install OpenCV with MSYS2:
                - pacman -S mingw-w64-x86_64-opencv
        - Install numpy and others in MSYS2
4. After all this, you should be able to create a python script and open both color and depth streams with gstreamer pipeline strings.

MartyG-RealSense, 2024-09-18:
It's no problem at all. I'm pleased to hear that you achieved a solution.  Thanks so much for sharing it!