realsenseai/librealsense #12645: Problems with USB 2.1 D455 unless I trigger Motion Module on/off

v-lopez, 2024-02-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.13.00.50| 
| Operating System & Version |   Ubuntu 20.04| 
| Kernel Version (Linux Only)    |  6.4.6                                         | 
| Platform                                 | PC|
| SDK Version                            |   2.50.0}                          | 
| Language                            |  C++                           | 
| Segment			| Robot                          | 

### Issue Description

I am connecting a D455 camera with a USB 2.1 cable, I am aware of the limitations.

This problems occurs both from my C++ application as well as the Realsense Viewer.

If I connect the camera and enable RGB or Stereo stream, I get "Frames didn't arrived within 5 seconds." error until I turn Motion Module ON and OFF and then the streaming starts.

[image]

MartyG-RealSense, 2024-02-09:
Hi @v-lopez  I note that in the top corner of your RealSense Viewer there is a red error box that states **RealSense UDEV rules are missing**.  The udev rules in Linux handle devices.

Which method did you use to install the librealsense SDK, please?

When installing the SDK from packages, the udev rules should be included in the **librealsense2-dkms** package.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

When compiling the SDK from source code, the udev rules can be installed by running the **./scripts/setup_udev_rules.sh** permissions script from the root  directory of the librealsense source code folder.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

v-lopez, 2024-02-09:
I built the SDK from sources.

I removed the camera, installed the udev rules with the script and reconnected them, but the behavior persists.

[image]

MartyG-RealSense, 2024-02-10:
Aside from this USB2 problem, if you plan to enable depth, RGB and color simultaneously in the Viewer then this is likely to trigger a different problem.  When these three streams are simultaneously enabled in the Viewer, the RGB or Depth stream usually becomes No Frames Received, even on a USB3 connection.  

There is no solution for that issue  in the Viewer other than to only enable depth + RGB or the Motion Module and depth or RGB (but not both).  The enabling of the Motion Module is the trigger for the problem, so changing the order that the streams are enabled in makes no difference.

The three-stream issue can be worked around in C++ though by using **callbacks** to access the IMU, like the SDK's **rs-callback** C++ example at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/examples/callback

MartyG-RealSense, 2024-02-10:
In regard to the USB2 issue, whilst I was not able to replicate your exact issue (depth + RGB do not enable until Motion Module is enabled), I did experience No Frames Received for depth and RGB on two different PCs on a USB2 connection.  I found that using a mains electricity powered USB hub enabled all of the streams to work correctly on USB2 on my own computers.

MartyG-RealSense, 2024-02-16:
Hi @v-lopez  Do you require further assistance with this case, please?  Thanks!

v-lopez, 2024-02-16:
Yeah, I still have the problem with the mains powered USB hub.

How can I do with code the equivalent of Turn ON and OFF the Motion Module?
At least I can workaround it by modifying the ROS driver.

MartyG-RealSense, 2024-02-16:
In code you can stop the pipeline, set a different list of **cfg** stream configuration instructions after the pipe.stop() instruction that do not include the accel and gyro streams and re-start the pipeline with the name of the amended configuration in the brackets to apply the configuration.

For example, start the program with `pipe.start (config1)` and then after setting a new config list called config2, do `pipe.start(config2)`

If the ROS2 wrapper is being used then accel and gyro streams can be enabled and disabled during runtime with a **ros2 param set** command.

```
ros2 param set /camera/camera enable_accel true
ros2 param set /camera/camera enable_gyro true
```

v-lopez, 2024-02-21:
Ok thank you.

closed 2024-02-21 by v-lopez