realsenseai/librealsense #12647: Getting Confidence Value for Pixel Point

ShankarKandasamy, 2024-02-09:
After going through this forum and pyrealsense2 docs; it's still not clear to me how a confidence value is obtained for a given pixel point.
I'm enabling the confidence stream like so:
`config.enable_stream(rs.stream.confidence, 1024, 768, rs.format.raw8, 30)`

I assume this stream would have to be aligned with the depth frame(?). And then how would I retrieve the confidence value for the depth at a given pixel location (x,y)?

I'm using Python. My end goal is to get the depth confidence value at a bunch of candidate points and then use the point with that has the highest confidence value.

Thanks in advance.

MartyG-RealSense, 2024-02-10:
Hi @ShankarKandasamy  The old RealSense SR300 camera model (which is now retired) was able to produce a confidence map but the 400 Series range of stereo depth cameras do not.  Another 400 Series user who attempted to access the confidence stream was provided advice at https://github.com/IntelRealSense/librealsense/issues/3185#issuecomment-460540961 regarding how they could achieve something similar to confidence mapping.

ShankarKandasamy, 2024-02-10:
Hi @MartyG-RealSense, I'm actually using the L515. Does this change anything?

MartyG-RealSense, 2024-02-10:
Yes, the confidence stream is usable on the L515 lidar depth model.  Advice provided at https://github.com/IntelRealSense/librealsense/issues/8655#issuecomment-809603271 states that the confidence stream of L515 is already perfectly aligned with the depth stream by default without you having to align them yourself.

ShankarKandasamy, 2024-02-10:
Thanks, @MartyG-RealSense. I was able to get the answer from  #8655 (comment). I'm not a C programmer so it took some struggle but below is the Python code that successfully prints the depth and corresponding confidence value when you left-click on a point on the depth map, incase it will be helpful to others. This code assumes the confidence stream of L515 is already aligned with the depth stream.
```
import pyrealsense2 as rs
import numpy as np
import cv2

# Mouse callback function to handle clicks in the depth window
def click_event(event, x, y, flags, params):
    if event == cv2.EVENT_LBUTTONDOWN:
        depth_frame = params.get('depth_frame')
        confidence_data = params.get('confidence_data')
        if depth_frame and confidence_data is not None:
            depth = depth_frame.get_distance(x, y)
            confidence_value = confidence_data[y, x]
            print(f"Clicked at ({x}, {y}) - Depth: {depth} meters, Confidence: {confidence_value}")

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.confidence, 640, 480, rs.format.raw8, 30)

try:
    pipeline.start(config)

    cv2.namedWindow('Depth Stream', cv2.WINDOW_AUTOSIZE)

    while True:
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        confidence_frame = frames.first(rs.stream.confidence).as_frame() if frames.first(rs.stream.confidence) else None

        if not depth_frame or not confidence_frame:
            continue

        depth_image = np.asanyarray(depth_frame.get_data())
        confidence_data = np.asanyarray(confidence_frame.get_data()) if confidence_frame else None

        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        params = {'Depth Frame': depth_frame, 'Confidence Data': confidence_data}

        cv2.setMouseCallback('Depth Stream', click_event, params)

        cv2.imshow('Depth Stream', depth_colormap)

        key = cv2.waitKey(1)
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break

finally:
    pipeline.stop()
```

closed 2024-02-10 by ShankarKandasamy

MartyG-RealSense, 2024-02-10:
You are very welcome.  I'm pleased to hear that you achieved a solution.  Thanks very much for the update and for sharing your solution!