realsenseai/librealsense #12648: Question about aligned depth to color and rs2_deproject_pixel_to_poin

Hechen0712, 2024-02-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.1| 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | PC  |
| SDK Version                            |   legacy / 2.54.2                         | 
| Language                            |  python                        | 
| Segment			| others                      | 

### Issue Description
Hi, I am trying to track the feature point using the D435i camera and have a problem.

I have aligned the depth frames to the color frames, and applied the post filtering, then load the depth and color interins(actually they are same after alignen, in this case is _1280x720  p[640.991 355.933]  f[916.63 917.641])_, and i selected the pixel_[x,y]_ of the feature point on the color frame and get the depth information of this point using: _filtered_frame.get_distance(int[x],int[y])_, and finally use: _point = rs.rs2_deproject_pixel_to_point(depth_intrin, x_y_float, depth_float)_ to back-project back to the world coordinates, and use the external parameter to transform the color coordinates to the depth coordinates: _point = rs.rs2_ transform_point_to_point(color_to_depth_extrin, point)_. 

But when I compare the data with the 3d point cloud in Intel Realsense Viewer, there are some difference in xyz coordinates respectively, where the xy gap is very small, only about 0.5mm, and the z-axis gap is the biggest, about 1.5mm-2mm(in this case, i measured: [0.01373m, 0.04220m,0.32808m]; In Viewer：[0.014m,0.043m,0.331m]), why is that? How can I get the accurate world coordinates?
Following is part of my code:

scale_deci = 2
filter_enable = 1

depth_to_disparity = rs.disparity_transform(True)
disparity_to_depth = rs.disparity_transform(False)
decimation = rs.decimation_filter()
spatial = rs.spatial_filter()
temporal = rs.temporal_filter()

decimation.set_option(rs.option.filter_magnitude, scale_deci)

temporal.set_option(rs.option.filter_smooth_alpha, 0.4)
temporal.set_option(rs.option.filter_smooth_delta, 20)

spatial.set_option(rs.option.holes_fill, 3)
spatial.set_option(rs.option.filter_magnitude, 3)
spatial.set_option(rs.option.filter_smooth_alpha, 0.30)
spatial.set_option(rs.option.filter_smooth_delta, 5)

align_to = rs.stream.color
align = rs.align(align_to)

try:
    while True:
frames = pipeline.wait_for_frames()
        decimated = decimation.process(frames).as_frameset()
        depth_frame = frames.get_depth_frame()
        aligned_frames = align.process(frames)
        color_frame = aligned_frames.get_color_frame()
        aligned_depth_frame = aligned_frames.get_depth_frame()

        if not color_frame or not aligned_depth_frame:
            continue

        filtered_depth_frame = depth_to_disparity.process(aligned_depth_frame)
        filtered_depth_frame = spatial.process(filtered_depth_frame)
        filtered_depth_frame = temporal.process(filtered_depth_frame)
        filtered_depth_frame = disparity_to_depth.process(filtered_depth_frame).as_depth_frame()
        for point in valid_points:
                      x, y = point.ravel()
                      cv2.circle(frame_to_draw, (int(x), int(y)), 2, color, -1)
                      x_y_float = [float(int(x)), float(int(y))]
                      if filter_enable:
                          depth = filtered_depth_frame.get_distance(int(x), int(y))
                      else:
                          depth = depth_frame.get_distance(int(x), int(y))

                      if depth == 0
                          continue
                      depth_float = float(depth)
                      point = rs.rs2_deproject_pixel_to_point(depth_intrin, x_y_float, depth_float)
                      point = rs.rs2_transform_point_to_point(color_to_depth_extrin, point)
                      coordinate_text = f"X:{float(point[0]):.5f} mm, Y:{float(point[1]):.5f} mm,Z:{float(point[2]):.5f} mm"
                      text_position = ((350, (60+i*20)))
                      cv2.putText(frame_to_draw, coordinate_text, text_position, cv2.FONT_HERSHEY_SIMPLEX, 0.6,
                                  color, 1)

MartyG-RealSense, 2024-02-11:
Hi @Hechen0712  If you are setting a 1280x720 resolution with the preset then that will not work, as when the pipeline starts it will automatically override the preset's resolution settings and apply the camera's default resolution.  On D435i the default is 848x480 depth and 1280x720 RGB.

There is a workaround that can be used though to read the resolution values from the json file and apply them to the pipe start.  This principle is demonstrated in a Python script at https://github.com/IntelRealSense/librealsense/issues/6075

Hechen0712, 2024-02-11:
> Hi @Hechen0712 If you are setting a 1280x720 resolution with the preset then that will not work, as when the pipeline starts it will automatically override the preset's resolution settings and apply [...]

Hi, thank you for the fast response, but unfortunately there seems to be no effect, the values in the z-direction still have an error of about 1.5-3mm.  Below is my measurement and point cloud generated in Viewer, is it because the depth frames are not fully aligned with the color frames? Or is it because I'm using rs2_transform_point_to_point incorrectly?
[image]

MartyG-RealSense, 2024-02-11:
Depth measurement error starts at around zero at the camera and increases linearly over distance as an observed object / surface becomes further from the camera.  This phenomenon is called **RMS Error**.  

So the measured distance typically will not exactly match the real-world distance and a Z-distance error of 1.5 to 2 mm is reasonable for an object that is at the distance of the one in the image above.

After depth to color alignment is performed with align_to, the color intrinsics should be used as the color sensor's center-line becomes the origin of depth after the alignment.

The point cloud of your Python script and that of the Viewer will not be exactly the same because the Python script aligns with align_to whilst the C++ based Viewer uses a different instruction called **map_to** to map depth and color together to construct a textured point cloud.  The map_to instruction is demonstrated by the C++ **rs-pointcloud** example program.

https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud

I also recommend that the **emitter_enabled** instruction is used and set to '1' so that the camera projects a dot patten onto the object to aid the camera's depth analysis.

`depth_sensor.set_option(rs.option.emitter_enabled, 1)`

Hechen0712, 2024-02-11:
Really appreciate your suggestion, my questions have all been answered.

MartyG-RealSense, 2024-02-11:
You are very welcome, @Hechen0712 - I'm pleased that I could help!

MartyG-RealSense, 2024-02-20:
Case closed due to all questions answered and no further comments received.

closed 2024-02-20 by MartyG-RealSense