realsenseai/librealsense #12653: Can't build RS-Kinfu

HasanTheSyrian, 2024-02-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   Ubuntu 22.04.3 LTS  | 
| Platform                                 | Apple Macintosh  |
| SDK Version                            | v2.54.2                          | 
| Language                            |  C++, OpenCV                        | 

----------------------------------------------------------------------------------------------------

This is the error I got when trying to make,

[image]

I was getting an error in the CMake GUI that the OpenCV dir is not set and that it was looking for `opencvconfig.cmake` so I added the path

[image]

[image]

Contents of: [OpenCVConfig.cmake](https://gist.github.com/HasanTheSyrian/61b6b589d544f7930e8be956b6fa1dcc) [CMakeCache.txt](https://gist.github.com/HasanTheSyrian/df54b31a512f5a0e2c1cc4093245b20d)

`modules` path
[image]

and `NONFREE` is selected as mentioned in the documentation

Docs used: [1](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu) [2](https://github.com/opencv/opencv_contrib/blob/master/README.md) [3](https://docs.opencv.org/4.x/d7/d9f/tutorial_linux_install.html)

After (im not 100% if i got this warning after failing to `make`) trying to configure/generate again I get this error in the CMake GUI for librealsense

```
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindOpenGL.cmake:315 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/aarch64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/aarch64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/aarch64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  wrappers/opencv/kinfu/CMakeLists.txt:7 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

Check for updates capability added to realsense-viewer
Check for updates capability added to realsense-depth-quality
Fetching recommended firmwares:
disable link time optimization for fw project
... https://librealsense.intel.com/Releases/RS4xx/FW/D4XX_FW_Image-5.15.1.0.bin
... https://librealsense.intel.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
... https://librealsense.intel.com/Releases/L5xx/FW/L51X_FW_Image-1.5.8.1.bin
Configuring done
```

HasanTheSyrian, 2024-02-11:
I commented out line 19 in https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/depth-filter/CMakeLists.txt and built successfully (have not tested yet)

closed 2024-02-11 by HasanTheSyrian

MartyG-RealSense, 2024-02-12:
It's great to hear that you achieved a successful build.  Thanks very much for the update!