realsenseai/librealsense #12654: Jetson Orin Jetpack 6.0 DP not detecting

Yuan-San, 2024-02-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  5.15                                         | 
| Platform                                 | NVIDIA Jetson AGX Orin  |
| Jetpack                                 | 6.0 Developer Version  |
| Jetson L4T                                 | 36.2  |
| SDK Version                            |  2.54.2                          | 
| Language                            |  Python                          | 

### Issue Description
Hello, I recently have access to the Intel Realsense D435i and trying to get the camera up and running using a Jetson AGX Orin. I tried building the SDK using CMake with `DFORCE_RSUSB_BACKEND` set to true and it outputs a smooth and successful building. But when I try running `realsense-viewer`, it fails to detect the camera. I also tried using the script `./libuvc_installation.sh` but it still returns nothing. Also tried the JetsonHacks installation method but I still got nothing. Here what the output when I try running realsense-viewer:

```
   12/02 16:24:35,692 INFO [281473192276000] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video0
   12/02 16:24:35,693 INFO [281473192276000] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video1
   12/02 16:24:35,693 INFO [281473192276000] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video2
   12/02 16:24:35,693 INFO [281473192276000] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video3
   12/02 16:24:35,694 INFO [281473192276000] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.3/video4linux/video4
   12/02 16:24:35,694 INFO [281473192276000] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.3/video4linux/video5
   12/02 16:24:35,705 WARNING [281473192276000] (d400-factory.cpp:1195) DS5 group_devices is empty.
```

`sudo dmesg`:
```
[11232.809292] usb 2-3.1: new SuperSpeed USB device number 6 using tegra-xusb
[11232.842183] uvcvideo 2-3.1:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[11232.842650] usb 2-3.1: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i  (8086:0b5c)
[11232.860298] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.1/2-3.1:1.0/input/input17
[11232.862540] uvcvideo 2-3.1:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[11232.862562] usb 2-3.1: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i  (8086:0b5c)
[11232.885375] hid-generic 0003:8086:0B5C.001E: device has no listeners, quitting
```

Does anyone know how to get my camera detected?

Yuan-San, 2024-02-12:
I noticed my issue is similar to #12566. Someone advises using `sudo` when running the realsense-viewer, but it still returns the same output.

MartyG-RealSense, 2024-02-12:
Hi @Yuan-San  Are you using the official 1 meter USB cable that is supplied with the camera or a USB cable of your own choice, please?

Does it make a difference if you unplug the micro sized end of the USB cable from the side of the camera, turn it around the opposite way and re-insert it (USB-C cables are two-way insertion at the micro-sized end)

Yuan-San, 2024-02-12:
Hello @MartyG-RealSense , Does it matter if I'm using the official USB cable or not? I am indeed using a USB cable of my own choice, but when I tested the depth camera using my own Windows laptop, the Intel.Realsense.Viewer.exe returned my camera and said it is connected using a 3.2 USB cable.

About flipping around the specified end of the cable, I tried it multiple times, but it didn't affect anything and the logs still returned the same thing.

MartyG-RealSense, 2024-02-12:
The cable can make a difference, because cables can be of varying quality.  RealSense requires a cable that is high quality and suited to transmitting high volumes of data, which is why an inexpensive cable such as a charging cable may be insufficient.

If a long cable is used (more than 2 meters) then the camera connection can degrade or disconnect if the cable is not a high quality one.  For cable lengths longer thn 5 meters, a cable with **active repeater** components in it to boost the signal strength over distance would be recommended.

If your chosen cable works normally with your Windows laptop though then it is likely okay nd not the cause of your issue.

Have you cleaned up your Jetson's storage and done a completely clean install of the JetsonHacks script?  If you have installed with CMake, libuvc and then JetSonHacks, you might have multiple conflicting versions of librealsense on your Jetson.

Yuan-San, 2024-02-12:
> The cable can make a difference because cables can be of varying quality. RealSense requires a cable that is high quality and suited to transmitting high volumes of data, which is why an inexpensive cable such as a charging cable may be insufficient.
> 
> If a long cable is used (more than 2 meters) then the camera connection can degrade or disconnect if the cable is not a high-quality one. For cable lengths longer than 5 meters, a cable with **active repeater** components in it to boost the signal strength over distance would be recommended.

I see, Thanks for clarifying! 

Well if there's it, I think I have no other choice but to clean and reinstall my Jetson. Was talking with my colleagues about the Jetpack 6 having little support too so I probably will also resort back to Jetpack 5.1. I will close this issue now. Thanks @MartyG-RealSense !

MartyG-RealSense, 2024-02-12:
You are very welcome!