realsenseai/librealsense #12656: RealSense T265 not recognized by RealSenseViewer

eagerlein, 2024-02-12:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|

My  RealSense T265 is not recognized by RealSenseViewer in several computers that I´ve tried.

toshNaik, 2024-02-28:
Yeah, I had the same problem with Ubuntu. Downgrading to 2.53.1 worked for me.

Yougo-robotics, 2024-11-13:
> Here is my solution for installing the latest version for T265 camera (that I know that works thus far). Assuming you have removed your librealsense via "sudo apt-get remove [input_library_name_nere]" Follow steps from top to bottom (I am using Ubuntu 22.04):
> 
> sudo apt-get install librealsense2=2.53.1-0~realsense0.8251
> 
> sudo apt-get install librealsense2-gl=2.53.1-0~realsense0.8251
> 
> sudo apt-get install librealsense2-net=2.53.1-0~realsense0.8251
> 
> sudo apt install librealsense2-dkms=1.3.22-0ubuntu1
> 
> sudo apt install librealsense2-utils=2.53.1-0~realsense0.8251

Very nice that is way simpler than recompiling the lib, works on t265 and ubuntu 22.04 kernel 6.11 thanks

66Lau, 2025-05-13:
Doesn't work in May 2025.

Need tp compile from source.
https://github.com/IntelRealSense/librealsense/issues/11843

closed 2026-10-23 by MartyG-RealSense