realsenseai/librealsense #12665: Depth image error with Intel RealSense Camera D435i

DAFS123, 2024-02-15:
Hi, I am having problems with the Intel RealSense D435i camera. I am trying to calibrate it with the "Dynamic Calibrator" application but it does not work. I updated the firmware to version 5.15.1 and the SDK to the latest version 2.54.2. When I use the application "Intel RealSense Viewer" the depth image appears like this: (The depth image appears with some black segments in the corners, this affects the measurement and I don't know where they came from.)
[image]

But when I activate the other sensors this appears:
[image]

I´m using Intel® RealSense™ SDK 2.0 (v2.54.2) and camera firmware 5.15.1.0.

I hope you can help me, my life is at stake in this project and the camera is not working.

Thank you

DAFS123, 2024-02-15:
Hi @MartyG-RealSense, thanks for your reply, I reset the camera hardware as you mentioned, I also reset the RealSense Viewer software and recalibrated with Dynamic Calibrator and it worked fine. However when testing the camera again it seems to be a bit displaced. Attached is a picture. I don't know why this is happening. 

I don't know if this is normal or if I should recalibrate something else. 

[image]

DAFS123, 2024-02-15:
I understand. I reran these exercises and this was the result.

Looking at the center of the camera:

[image]

Raising hand to place more things on the scene.:

[image]

MartyG-RealSense, 2024-02-15:
This silhoutte around the arm / hand is normal.  Of all of the body parts, the black outline is most likely to occur around the arm and hand when holding it up to the camera.  The silhoutte should shrink in width when the arm / hand is moved away from the camera and become larger when moved closer to the camera.

Changing the **Depth Units** of the camera from the default '0.001' to the smaller '0.0001' value may sometimes improve the hand / arm silhoutte, though not completely remove it.

[image]

DAFS123, 2024-02-15:
Thank you very much @MartyG-RealSense , the problem has been solved using your help. https://github.com/IntelRealSense/librealsense/issues/12665#issuecomment-1946059215

closed 2024-02-15 by DAFS123