realsenseai/librealsense #12666: Realsene D455 firmware update gone wrong, Unable to detect

Jaydeepbais, 2024-02-15:
tried to install different firmware and now the camera is not detecting in realsense viewer.
on Device Manager it says Unknown USB Device (invalid Configuration Descriptor). How can i update my firm ware to the latest one again. Or reset my device to default firmware? rs-fw-update -l while running this command, I got this error "There are no connected device"

MartyG-RealSense, 2024-02-15:
Hi @Jaydeepbais  On another RealSense support forum where you commented about this problem, you stated that you had disconnected the camera depth module so that only the Vision Processor D4 V3 board was connected to USB and you shared your Windows Device Manager image.  This showed the camera to be detected as **405i Tester**.  This ID is normal for a D455 that has had its camera module disconnected as it is only detecting the D4 V3 board.

[image]

As '405i Tester' is not recognized by RealSense as a valid camera model ID, the firmware is unable to be updated whilst the camera module is disconnected.

I note that on an image of the RealSense Viewer that you shared on the other forum site, your Viewer version is **2.44.0**.

[image]

The recommended firmware for the 2.44.0 version is **5.12.12.100**.  Do you recall which firmware version you attempted to update the camera to, please?

After reconnecting the camera board, is the camera able to be detected if you download and run the **Intel.RealSense.Viewer.exe** file from the 'Assets' list of the 2.54.2 release at the link below, please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

[image]

Jaydeepbais, 2024-02-15:
Hi MartyG, Recently I tried disconnecting the internal FPC and then connecting to PC. This method also didn't worked for me Device Manager is showing Intel RealSense Depth Camera 405i Tester after removing FPC. And I have attached the screenshot of the error i am getting in Intel Realsense Viewer.
Please suggest me the way to reinstall the firmware in this situation.
Thankyou
[image]
[image]

Jaydeepbais, 2024-02-15:
Hi MartyG, the firmware that installed  is 5.12.12.100.

MartyG-RealSense, 2024-02-15:
The minimum firmware that supports D455 is now **5.13.0.50**.  So the 5.12.12.100 firmware will not know how to handle a D455.

[image]

This is an issue that is not fixable with a firmware update by the user, unfortunately.  I will consult with my Intel RealSense colleagues.

May I ask which country you are located in, please?

Jaydeepbais, 2024-02-15:
I am located in Mumbai, India. If you could help me it would be great
Thank you

Jaydeepbais, 2024-02-15:
Hi MartyG, I have read that you can reinstall the firmware if you have any older d4 board.... is it true ?

MartyG-RealSense, 2024-02-15:
You can, though it would be difficult to do so unless you already own an older D4 V3 board (it has to be the V3 version).  There is a sticker on the D4 board with a version number, but you would have no guarantee of getting the correct board version number from a retailer unless they could check the sticker themselves before shipping it to you.

Jaydeepbais, 2024-02-15:
Is there any other way of installing the correct firmware without older D4 board?

MartyG-RealSense, 2024-02-15:
No, the only other method involves my Intel RealSense colleagues, whom I am communicating with on your behalf.

Jaydeepbais, 2024-02-15:
Ok thank you for your help, let me know

Jaydeepbais, 2024-02-15:
Hi MartyG, I am able to buy couple of d450 boards in india, can u suggest which one of them should i buy to update the firmware
[https://www.mouser.in/ProductDetail/Intel/82635DSASICBDIF?qs=hd1VzrDQEGgFDJnfCF8Xhg%3D%3D](url)
[https://www.mouser.in/ProductDetail/Intel/82635DSASMDLPRQ?qs=wd5RIQLrsJhT7QD5H29iow%3D%3D](url)

Thankyou

MartyG-RealSense, 2024-02-15:
The above links are both for the D4 V3 board rather than the D450 module.  The D450 module is at the link below.

https://www.mouser.in/ProductDetail/Intel/82635DSD450?qs=hd1VzrDQEGjg%252BAsSbWbygA%3D%3D

It has the product code **999WCM**.  However, the older D450 board version **K83121-100** and the newer version **K83121-101 both have the same product code, so that code will not be a helpful reference for distinguishing between the versions and the store images do not show the version information sticker.

[image]

MartyG-RealSense, 2024-02-16:
Hi @Jaydeepbais  Please send me an email at the address below where I can respond with further advice.

martyx.grover@intel.com

MartyG-RealSense, 2024-05-18:
Hi @Jaydeepbais  Can this issue now be closed?  Thanks!

closed 2024-05-26 by MartyG-RealSense