realsenseai/librealsense #12669: Issue converting Bag file to infrared video

OliviaNocentini, 2024-02-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {d435i} | 
| Firmware Version                   | (Open RealSense Viewer --> Ubuntu) | 
| Operating System & Version |   {Ubuntu 20.04] | 
| Language                            |  {Opencv}                          | 
| Segment			|  {Camera}                          | 

### Issue Description
Hi all, this question has been asked many times but I cannot find a proper answer. I store a bag file using realsense-viewer with the RGB and infrared topics. When I try to convert the bag file into two differen videos for the RGB and infrared  I can get the video with the RGB stuff but I cannot open the video of the infrared. Here the code I implemented

`import argparse
import os
import rospy
import cv2
import rosbag
from sensor_msgs.msg import Image
from cv_bridge import CvBridge

def bag_to_video(bag_file, output_video_1, output_video_2):
    
    rospy.init_node("bag_to_video")
    
  
    bridge = CvBridge()

  
    bag = rosbag.Bag(bag_file, "r")

    
    fourcc = cv2.VideoWriter_fourcc(*'mp4v')
    
    
    video_writer_1 = None
    video_writer_2 = None

    # Iterate through bag messages
    for topic, msg, t in bag.read_messages():
        if topic == '/device_0/sensor_1/Color_0/image/data':
            # Convert ROS Image message to OpenCV image
            cv_image = bridge.imgmsg_to_cv2(msg, desired_encoding="bgr8")

            # Initialize video writer for the first topic
            if video_writer_1 is None:
                height, width, _ = cv_image.shape
                video_writer_1 = cv2.VideoWriter(output_video_1, fourcc, 30, (width, height))

            # Write frame to video for the first topic
            video_writer_1.write(cv_image)
            
        elif topic == '/device_0/sensor_0/Infrared_2/image/data':
            # Convert ROS Image message to OpenCV image
            cv_image = bridge.imgmsg_to_cv2(msg, desired_encoding="passthrough")

            # Initialize video writer for the second topic
            if video_writer_2 is None:
                height, width = cv_image.shape[:2]
                video_writer_2 = cv2.VideoWriter(output_video_2, cv2.VideoWriter_fourcc(*'mp4v'), 30, (width, height))

            # Write frame to video for the second topic
            video_writer_2.write(cv_image)

    # Release video writers and close bag file
    if video_writer_1 is not None:
        video_writer_1.release()
    if video_writer_2 is not None:
        video_writer_2.release()
    bag.close()

if __name__ == "__main__":
    # Parse command-line arguments
    parser = argparse.ArgumentParser(description='Convert ROS bag file to video')
    parser.add_argument('bag_file', type=str, help='Path to the input bag file')
    parser.add_argument('output_folder', type=str, help='Folder where the output videos will be saved')
    args = parser.parse_args()

    # Create the output folder if it does not exist
    if not os.path.exists(args.output_folder):
        os.makedirs(args.output_folder)

    # Define output video file paths for each topic
    output_video_1 = os.path.join(args.output_folder, "output_video_topic1.mp4")
    output_video_2 = os.path.join(args.output_folder, "output_video_topic2.mp4")

    # Convert bag file to video for each topic
    bag_to_video(args.bag_file, output_video_1, output_video_2)`

Thank you for yor answer.

OliviaNocentini, 2024-02-15:
I solved the issue changing the following line and now it works!
`video_writer_2 = cv2.VideoWriter(output_video_2, cv2.VideoWriter_fourcc(*'mp4v'), 30, (width, height))`

MartyG-RealSense, 2024-02-15:
Hi @OliviaNocentini  Thanks so much for sharing your script and the solution that worked for you!

OliviaNocentini, 2024-02-15:
I have another question,now I have the infrared and the RGB videos but the videos appear to have different length in seconds even if they belongs to the same data aqcuisition. How is it possible?Is there a way to say to synchronize the two topics  in order to obtain videos of the same length?

MartyG-RealSense, 2024-02-15:
Do you have the option to record your bag in the RealSense ROS wrapper with **rosbag record** instead of recording in realsense-viewer, please?  If you are able to record in the wrapper then you could launch it with the parameter **enable_sync:=true** included in the roslaunch instruction in order to gather the closest frames of different sensors (infrared, color and depth) to be sent with the same timetag.

OliviaNocentini, 2024-02-15:
The problem is that I already record the bag using realsense-viewer and not using rosbag record. Is there any other option to obatin sync video after having recorded the bag using realsense-viewer?

MartyG-RealSense, 2024-02-15:
A problem with recording a bag in realsense-viewer is that the Stereo Module and RGB streams have to be enabled separately and so whichever stream type was started first will have frame numbers that are ahead of the other as it will take at least a second to enable both streams individually.

An example of this is shown below, where I started both streams in 1 second with fast clicking but one stream had a frame number of 483 and the other was 486.

[image]

The same issue can be seen in one of the RealSense SDK's official sample bags, where at the point in playback that I paused, depth is at frame number 2669 and RGB is at frame 2567.

[image]

There are not any controls in realsense-viewer to sync up the streams during playback after the bag has been recorded.  Enabling the depth and RGB streams within 1 second with quick clicks may be the best that can be done.