realsenseai/librealsense #12673: librealsense.github.io yields a 404

hernot, 2024-02-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   not relevant | 
| Firmware Version                   | not relevant | 
| Operating System & Version |  not relevant | 
| Kernel Version (Linux Only)    | not relevant                               | 
| Platform                                 | not relevant |
| SDK Version                            |  librealsense2                          | 
| Language                            |  all                         | 
| Segment			|  not relevant                   | 
| Documentation         | librealsense.github.io  |

### Issue Description
any link leading to documentation on https://librealsense.github.io yields a 404 http error. Seems as if the links would either not be up to date, the last documentation build did not complete sucessfully leaving corrupt pages or the service is down. Just tested.

MartyG-RealSense, 2024-02-15:
Hi @hernot  The `https://intelrealsense.github.io/` address that contains the official API documentation is also experiencing the same 404 issue at the time of writing this.  These addresses may return on their own later.

In the meantime, there is a version of the documentation at the link below that draws directly from the official API documentation and is not currently affected by the 404.

https://unanancyowen.github.io/librealsense2_apireference/classes.html

hernot, 2024-02-19:
@MartyG-RealSense Thank you very much, but as of today the https://intelrealsense.github.io/ pages still show the 404 state that there are no pages. Possibly on the last build something went unnoticed or false classified (warning instead of error) severely wrong. 

[image]

MartyG-RealSense, 2024-02-19:
Thanks very much for reporting this issue again.  I have highlighted it to my Intel RealSense colleagues.

Z14-A, 2024-03-02:
trying to navigate to intelrealsense github docs page still gives 404 error.
[image]

Is there any other documentation on pyrealsense2 library out there? I tried building the docs locally but that is generating just an empty template.

Z14-A, 2024-03-02:
Trying to navigate to the intelreasense github page still gives 404 error. 

Is there any other documentation on pyrealsense2 wrapper out there? I tried building the docs locally but that generates just an empty template.

MartyG-RealSense, 2024-03-02:
The pyrealsense2 documentation is currently offline due to the 404 issue.  However, you can use the **Wayback Machine** internet page archiving site to access a version of the documentation from 2023.

https://web.archive.org/web/20230328071532/https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.html

MartyG-RealSense, 2024-03-21:
Hi @Z14-A  Was the Wayback Machine documentation link above helpful for you, please?  Thanks!

Z14-A, 2024-03-21:
Yes, wayback machine was helpful. Got references to all the classes and corresponding functions and attributes. Nevertheless, it still is a fallback option. Would highly appreciate pointers to creating offline documentation locally.
Thank you.

MartyG-RealSense, 2024-03-21:
Whilst it is possible to build librealsense documentation locally from its downloadable source code using **doxygen**, I think that is just going to install documentation relevant to the C++ API rather than the pyrealsense2 wrapper.  Though pyrealsense2 instruction names are typically the same as the C++ ones except that the names start with **rs** instead of **rs2**.

https://github.com/IntelRealSense/librealsense/blob/master/doc/readme.md#documentation

MartyG-RealSense, 2024-05-30:
Hi @hernot and @Z14-A  The C++ API docs site is now restored.  Work is continuing on restoration of the pyrealsense2 site.  Thanks so much for your patience!

MartyG-RealSense, 2024-07-18:
Hi everyone, the pyrealsense2 docs site is now restored.

https://intelrealsense.github.io/librealsense/python_docs/

MartyG-RealSense, 2024-07-24:
Case closed due to documentation sites having been restored.

closed 2024-07-24 by MartyG-RealSense