realsenseai/librealsense #12674: [L515 sensor] No "color" frame with sample SDK exmaple,  while default  viewer (Intel.RealSense.Viewer.exe) shows "color" frame well.

JeongJunLee, 2024-02-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.8.1 | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    | N/A                                       | 
| Platform                                 | PC  |
| SDK Version                            | Intel.RealSense.SDK-WIN10-2.54.2.5684                     | 
| Language                            |  C/C++                        | 
| Segment			|  Robot                         | 

I connected the device (L515) into USB3.0 correctly. 
With Intel.RealSense.Viewer.exe, I can successfully preview all images (color, depth, ir..). 
But when I execute sample SDK app without modifying any source code, 
there is NO valid color frame, but only ir frame!!!
```
    pipe.start();
    auto color = frames.get_color_frame();
    if (!color)
        color = frames.get_infrared_frame(); // <-- comes here
```

And if I try starting pipe using configuration with "color stream",  pipe.start() fails with exception. 
```
    configmy.enable_stream(RS2_STREAM_DEPTH, 1024, 768);
    configmy.enable_stream(RS2_STREAM_CONFIDENCE);
    configmy.enable_stream(RS2_STREAM_INFRARED);    
    configmy.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_RGB8);

    profile = pipe.start(configmy); // <-- crashes with exception!
```

I suppose there are some "error handling codes" inside Intel.RealSense.Viewer.exe. 

Anyway, please can you help me with this problem?

JeongJunLee, 2024-02-16:
p.s. 
One more weired thing is that some PC's do not show above error!!! 
I found 2 PC's which show above error, and also other 2 PC's which runs well.

MartyG-RealSense, 2024-02-16:
Hi @JeongJunLee  Could you first try unplugging the micro-sized end of the USB cable from the base of the camera, reversing the connector's orientation and plugging it back into the L515 camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing this make any difference?

What is the name of the sample SDK app that you are using, please?

JeongJunLee, 2024-02-16:
Hello MartyG, 

Thank  you for quick reply. 

"Unplugging and reversing the orientation and re-plugging the micro-sized end of USB cable to the camera"
made NO difference. 

But, "Unplugging and re-plugging the PC side USB port (type A)" made difference.!! 
After re-plugging PC side USB port, it works fine!!! 
Both default viewer and sample SDK app works well showing color frame!!!

Immediate problem is solved. 
But I am afraid in the future the same problem might re-happen. 

Do you know the root cause of it, and could you give us any solution which solves this issue fundamentely? 
I would really appreciate it. 
Thank  you for your support.

MartyG-RealSense, 2024-02-16:
If the problem is at the full-size USB-A port rather than the micro-sized USB-C port then you could try resetting the camera when the program is run with a **hardware_reset()** instruction, like in the C++ code example for doing so at https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

JeongJunLee, 2024-02-19:
Hi MartyG, 

Thank you for your advice. 
But, it doesn't help.
Even after reseting with below code, the situation is still all the same. 
Maybe other way might be needed..
```
rs2::context ctx;
rs2::device dev = ctx.query_devices().front(); // Reset the first device
dev.hardware_reset();
rs2::device_hub hub(ctx);
dev = hub.wait_for_device();
```

MartyG-RealSense, 2024-02-19:
When you run your script, are there any other programs already active that are already using the camera?  Your script should be the only RealSense program that is running.

JeongJunLee, 2024-02-19:
No other programs are active. 
My app is the only active program. 
So, the IR image is well captured (The problem is that "color" image is not available).

MartyG-RealSense, 2024-02-19:
You mentioned at the beginning of this case that you tested a sample SDK app without modifying its source code.  Which app did you test, please?  

Does the color image appear if you test the C++ **rs-capture** app please?  If you have installed the full version of the RealSense SDK using the **Intel.RealSense.SDK-WIN10** installer program then you can find a pre-built executable version - **rs-capture.exe** in the SDK's 'Tools' folder at `C:\Program Files (x86)\Intel RealSense SDK 2.0\tools` that can be launched by double-leftclicking on the .exe.

JeongJunLee, 2024-02-20:
What I used was "rs-pointcloud". 

I tested the pre-built apps of both "rs-pointcloud.exe" and  "rs-capture.exe" in `C:\Program Files (x86)\Intel RealSense SDK 2.0\tools`. 
But the result was all the same. 
(Intel.RealSense.Viewer.exe shows color images well, but 
pre-built apps of both "rs-pointcloud.exe" and  "rs-capture.exe" show only IR image.) 

During a few days, I tried to reproduce the error and 
found something and want to share with you, just for your information. 
1. I tried to reproduce the error but it was not easily reproduced. 
    I reproduced it only 3 times in one computer. 
    I found 100% reproductivity way: It is to plug the PC-side USB end (type A) to the PC pretty "slowly". 
2. Once the error happens, hardware_reset() does not help. 
3. Even at the error situation, the default viewer always shows color images well.

Most important fact is
4. When the error happens, unpluging and repluging of the PC-side USB to the PC solves the problem, almost always. 

So I feel that I found a solution (#4 above) anyway.  :) 
(I wonder how the default viewer always works well, but the above solution (#4) is enough for me now.) 

I really appreciate your time and support. 
If you find any further information or tips and let us know, it would also be much appreciated. 

Anyway, I think this issue is resolved and can be closed. 
Thank you for your help. 
Sincerely.

MartyG-RealSense, 2024-02-20:
You are very welcome.  Thanks so much @JeongJunLee for the detailed feedback.  As you suggested, I will close the issue.  Thanks again!

closed 2024-02-20 by MartyG-RealSense