realsenseai/librealsense #12675: How to build and use rs-sensor-control.cpp?

IAMHAADICOOL, 2024-02-16:
Can anyone please explain how to build and  use the methods defined in rs-sensor-control.cpp?

MartyG-RealSense, 2024-02-16:
Hi @IAMHAADICOOL  Example programs in the RealSense SDK are built in a different way to scripts created by RealSense users.  The examples are first built as a set of programs during installation of the librealsense SDK by including the flag **-DBUILD_EXAMPLES=TRUE** in the CMake build instruction, followed by the **make** instruction.

If one of the example programs is modified and the 'make' instruction input again, only the examples that have been modified are built and not the entire set of examples.  More information about this can be found at https://github.com/IntelRealSense/librealsense/issues/2946 and https://github.com/IntelRealSense/librealsense/issues/6797

It is possible to copy the code from an SDK example such as rs-sensor-control into your own project but you typically end up with missing details in the CMakeLists.txt file that prevent the librealsense libraries from being found.  https://github.com/IntelRealSense/librealsense/issues/10715#issuecomment-1193351569 has advice about creating a CMakeLists.txt file with the necessary linkages.

IAMHAADICOOL, 2024-02-16:
Thanks for help @MartyG-RealSense 😄

MartyG-RealSense, 2024-02-23:
Thanks very much for the update.  As you suggested, I will close the issue.  Thanks again!