realsenseai/librealsense #12677: Aligning external camera and depth in Python using numpy arrays results in black image expect the first pixel

fcitil, 2024-02-17:
| Required Info                               |                                                   |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Operating System & Version |   Ubuntu 20.04.6 LTS  | 
| Kernel Version (Linux Only)    |  5.15.0-94-generic                                       | 
| Platform                                 | PC  |
| SDK Version                            |  2.53.1.4623                       | 
| Language                            |  Python 3.9.16                          | 

### Issue Description
Hi,
  I am trying to align a external RGB camera pixels with Realsense D435i depth usign align.process. In order to achieve that, I am trying to use `software-device` with two seperate rostopic of RGB frame and depth topics with ros. In this regard, I am currently working with .npy files to confirm that I am able create a new frameset consisting of RGB frame and corresponding depth.  I first tried to run [ sample python script](https://github.com/IntelRealSense/librealsense/issues/12020#issuecomment-1675211260) provided by @gwen2018 in issue [#12020](https://github.com/IntelRealSense/librealsense/issues/12020), but I need to change the first part in order to work with realsense .bag ( I used official [sample](https://librealsense.intel.com/rs-tests/TestData/d435i_sample_data.zip) .bag files) file as follows:
  ```
  #####################################################################################################
#   Demo                                                                                           ##
#     Align depth to color with precaptured images in software device                              ##
#                                                                                                  ##
##  Purpose                                                                                        ##
##    This example first captures depth and color images from realsense camera and then            ##
##    demonstrate align depth to color with the precaptured images in software device              ##
##                                                                                                 ##
##  Steps:                                                                                         ##
##    1) stream realsense camera with depth 640x480@30fps and color 1280x720@30fps                 ##
##    2) capture camera depth and color intrinsics and extrinsics                                  ##
##    3) capture depth and color images and save into files in npy format                          ##
##    4) construct software device from the saved intrinsics, extrinsics, depth and color images   ##
##    5) align the precaptured depth image to to color image                                       ##
##                                                                                                 ##
#####################################################################################################

import cv2
import pyrealsense2 as rs
import numpy as np
import os
import time

fps = 30                  # frame rate
tv = 1000.0 / fps         # time interval between frames in miliseconds

max_num_frames  = 300    # max number of framesets to be captured into npy files and processed with software device

depth_file_name = "depth"  # depth_file_name + str(i) + ".npy"
color_file_name = "color"  # color_file_name + str(i) + ".npy"

# intrinsic and extrinsic from the camera
camera_depth_intrinsics          = rs.intrinsics()  # camera depth intrinsics
camera_color_intrinsics          = rs.intrinsics()  # camera color intrinsics
camera_depth_to_color_extrinsics = rs.extrinsics()  # camera depth to color extrinsics

######################## Start of first part - capture images from live device #######################################
# stream depth and color on attached realsnese camera and save depth and color frames into files with npy format
try:
    # # create a context object, this object owns the handles to all connected realsense devices
    ctx = rs.context()
    devs = list(ctx.query_devices())
    
    if len(devs) > 0:
        print("Devices: {}".format(devs))
    else:
        print("No camera detected. Please connect a realsense camera and try again.")
        # exit(0) # since no device is needed to be work with .bag files skip this line
    
    pipeline = rs.pipeline()

    # configure streams
    config = rs.config()
    config.enable_stream(rs.stream.depth,1280, 720)#, rs.format.z16, fps)  # when formats and fps added
    config.enable_stream(rs.stream.color, 640, 480)#, rs.format.bgr8, fps)   # It fails, I don't know why ??
    
    config.enable_device_from_file("d435i_walking.bag") # add this line to work with .bag file
[... 234 lines cut ...]

    if key == 13 or key == 32: paused = not paused
        
    if paused:
        print("Paused for 5 seconds ...", i, ", press ENTER or SPACEBAR key anytime for additional pauses.")
        time.sleep(5)
        paused = not paused

# end of second part - align depth to color with the precaptured images in software device
######################## End of second part - align depth to color in software device #############################
    
cv2.destroyAllWindows()
 ```
Although, the first part seems to properly saves .npy files and I can get frameset with color and depth image using this files, the align.process give black frame, except the [0,0] indexed pixel as in issue [#6522](https://github.com/IntelRealSense/librealsense/issues/6522#issuecomment-639629549). Therefore, I applied [the solution](https://github.com/IntelRealSense/librealsense/issues/6522#issuecomment-640643730) stated in this issue, though it is given in C++, by adding the following lines to the python script:
 ```
 # add read only option to the depth sensor
depth_sensor.add_read_only_option(rs.option.depth_units, 0.001)  # 0.001 meters
depth_sensor.add_read_only_option(rs.option.stereo_baseline, 49.93613815307617)  # 0.001 meters
 ```
but still I get black image, except the first pixel, for aligned depth image. Since the first part of the [sample python script](https://github.com/IntelRealSense/librealsense/issues/12020#issuecomment-1675211260) seems (we cannot explicitly state fps and data format for config) working and I created required .npy files thanks to it, I ended up using [this script](https://github.com/fcitil/Event-Depth-Fusion/blob/realsense_pipeline/software_device_demo.py) that includes only the second part of the [sample python script](https://github.com/IntelRealSense/librealsense/issues/12020#issuecomment-1675211260) and required read_only_options additions given above.

  Is it possible that, the problem with align process might be due to the fact that the fps and encoding cannot be stated explicitly in the first part, which would cause the problem when using the corresponding numpy arrays? If not what could be the problem with align.process considering that given scripts are able to show unaligned depth and color images from frameset created by `software-device` ? I couldn't find any working reference for this problem, any advice would be appreciated.
 
 Thank you in advance,
 Furkan Çitil

MartyG-RealSense, 2024-02-18:
Hi @fcitil  As only the first pixel is being aligned, that makes me wonder whether this is because the program is not looping through the frames, like in the **align_depth2color.py** alignment example.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py#L63

fcitil, 2024-02-18:
Hi @MartyG-RealSense, I am able to run align-depth2color.py script with the official sample realsense .bag file successfully. It would have been the case, if the program is not able to show unaligned depth and color frames in the loop. I think, the problem is most likely to be related to use of `software-device` with `align.process`. Instead of the loop in the python script for frames, is it possible that the cpp program for align.process in SDK is not able to loop through the frames?

MartyG-RealSense, 2024-02-18:
The software-device interface has rarely worked with Python without errors and is best used with C++.  There are therefore very few working examples of Python code for software-device, which makes it difficult to diagnose any possible problems in your script that are preventing alignment from working.

Does alignment take place if the sample Python script at https://github.com/IntelRealSense/librealsense/issues/12020#issuecomment-1675211260 is run on your computer?

fcitil, 2024-02-18:
As I stated in the [description](https://github.com/IntelRealSense/librealsense/issues/12677#issue-2140633452), it does not work with .bag files directly. I needed to change and add following lines
```
  # configure streams
  config = rs.config()
  config.enable_stream(rs.stream.depth,1280, 720)#, rs.format.z16, fps)  # when formats and fps added
  config.enable_stream(rs.stream.color, 640, 480)#, rs.format.bgr8, fps)   # It fails, I don't know why ??
  
  config.enable_device_from_file("d435i_walking.bag") # add this line to work with .bag file
  ```
  And I get black image as an aligned depth frame, except the first pixel.

MartyG-RealSense, 2024-02-18:
What happems if you comment out `config.enable_device_from_file` and have a D435i attached so that the script tries to use a live camera instead of a bag file?

fcitil, 2024-02-18:
I haven't tried yet with the camera have attached. Let me try it with D435i in the following days. However, I am trying to align depth stream of realsense camera and color stream of an external camera. Therefore, I need to work with numpy arrays as [here](https://github.com/fcitil/Event-Depth-Fusion/blob/realsense_pipeline/software_device_demo.py), which will end up with the same problem even if I use the D435i, so I don't think having attached D435 would solve the problem with `align.process` when working with numpy arrays.
  Eventually, I am planning to get depth frames and color frames from external camera using ros and align numpy frames as in the sample python script. Do you have better suggestion for this purpose?

MartyG-RealSense, 2024-02-18:
Is use of numpy compulsory for your project, please?  The link below has an Intel guide for using ROS1 to combine the views of two separate cameras together.

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

fcitil, 2024-02-18:
I decided to use numpy since the sample-python script saves .npy files and use numpy arrays to create frameset. However, it is not necessary to use numpy and if there is another solution to align external camera with different format, it would be also suitable for me. The guide that you provided seems to be suitable for two realsense camera with serial numbers and it explains combining point clouds. However, I am trying to align RGB frame of external non-realsense camera whose relative pose w.r.t. realsense camera is known and realsense depth frame as in the official **align_depth2color.py** alignment example.

fcitil, 2024-02-19:
> What happems if you comment out the `config.enable_device_from_file` line and have a D435i attached so that the script tries to use a live camera instead of a bag file?

I have just tried with D435i camera have attached, still I got the same response. I get color and depth frames separately but I don't get aligned frame as a result of align.process.

MartyG-RealSense, 2024-02-19:
I conducted extensive further research into this issue but https://github.com/IntelRealSense/librealsense/issues/12020 is the best available reference unfortunately and there does not appear to be any other approach available to using software-device with alignment in Python.  I do apologize.

MartyG-RealSense, 2024-02-27:
Hi @fcitil  Bearing in mind the information in the comment above, do you require further assistance with this case please?  Thanks!

fcitil, 2024-02-27:
I have searched through all related issues but have not found any solution for my problem. I have tried to run the [script](https://github.com/IntelRealSense/librealsense/issues/12020#issuecomment-1675211260) with the sdk version that is said to be working in the issue, but it is not. I am still investigating for a possible solution ...

MartyG-RealSense, 2024-02-27:
Please do share a solution here if you find it.  Good luck!

MartyG-RealSense, 2024-05-18:
Hi @fcitil  Do you require further assistance on this case please as your last comment was in February 2024.  Thanks!

fcitil, 2024-05-19:
I am still not able to use software-device in python, I implemented the code for pixel matching for an external camera and D435i depth camera, which performs de-projection from depth image to depth camera coordinates, transformation from depth camera coordinates to external camera coordinates, and then projection to external camera for each depth pixel. However, it is too slow for realtime applications since it performs pixel-wise sequential operation with my current implementation without utilizing realsense pipeline, unfortunately.

MartyG-RealSense, 2024-05-19:
I researched your case again from the beginning but did not find a better solution than the software-device script at https://github.com/IntelRealSense/librealsense/issues/12677 that you have already looked at, unfortunately.

The current latest librealsense version 2.55.1 has implemented a fix for software-device's matcher creation - as described at https://github.com/IntelRealSense/librealsense/pull/12394 - so it may be worth testing software-device again in 2.55.1 if you have not done so already.

MartyG-RealSense, 2024-05-26:
Hi @fcitil  Have you tried testing software-device in SDK version 2.55.1 please?

MartyG-RealSense, 2024-06-03:
Thanks very much for the update.  Good luck!

Irene-G, 2024-06-27:
Hi, I had the same problem as @fcitil and just tested the code with the latest version of the SDK (2.55.1.6486). While this version has resolved the initial issue, I still don't get the expected results. In my experiments, I recorded a .bag file using the D455 camera and then exported the images from the .bag, both with Depth aligned to RGB and without alignment. I used the images from the latter case to align them using `software-device` and compared the results with those obtained by aligning directly from the .bag, getting different results.

[image]
*Figure 1: Image without alignment*

[image]
*Figure 2: Aligned image from the .bag file*

[image]
*Figure 3: Image aligned using `software-device`*

I don't know if I'm doing something wrong, but I get the impression that the extrinsic values that are being used are not the ones I'm setting them to, which are the values read from the .bag:

```
depth_to_color_extrinsics.rotation = [0.99998, -0.00513663, -0.00364226, 0.00512824, 0.999984, -0.00230822, 0.00365406, 0.00228949, 0.999991]
depth_to_color_extrinsics.translation = [-0.0588909, 0.000456639, 0.000275651]
color_to_depth_extrinsics.rotation = [0.99998, 0.00512824, 0.00365406, -0.00513663, 0.999984, 0.00228949, -0.00364226, -0.00230822, 0.999991]
color_to_depth_extrinsics.translation = [0.0588931, -0.000153989, -6.15026e-05]

depth_profile.register_extrinsics_to(depth_profile, depth_to_color_extrinsics)
color_profile.register_extrinsics_to(color_profile, color_to_depth_extrinsics)
```
What do you think about it?

Thanks in advance for your help!

MartyG-RealSense, 2024-06-28:
Hi @Irene-G  The image aligned using software-device appears to be the same as the first image which did not have alignment.

https://github.com/IntelRealSense/librealsense/issues/12020#issuecomment-1675211260 - which the original poster on this case @fcitil highlighed - remains the best available reference for using intrinsics and extrinsics with software-device for the purposes of depth-color alignment.  I do not have further advice on the topic, unfortunately.

MartyG-RealSense, 2024-07-04:
Hi @fcitil  and @Irene-G  Do either of you have updates about this case that you can provide, please?  Thanks!

closed 2024-07-18 by MartyG-RealSense