realsenseai/librealsense #12684: combine realsense d435 with lidar l515

laracb01, 2024-02-19:
I am trying to combine the use of my REALSENSE DEPTH CAMERA D435 and my LIDAR CAMERA L515. When uploading both devices to the INTEL REALSENSE VIEWER, I am unable to combine the visualization of both obtained datas. 

| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|008222071408|
|**Firmware Version**|5.12.10|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|008523020141|
|**Firmware Update Id**|008523020141|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0190890|
|**Firmware Version**|1.3.15.100|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003a9650abb|
|**Firmware Update Id**|0003a9650abb|

Please provide a description of the problem

MartyG-RealSense, 2024-02-19:
Hi @laracb01  The RealSense Viewer tool can combine streams from different cameras together in its 3D pointcloud mode after you have added the second camera to the side-panel using the **Add Source** button.  

However, you may need to disable the Viewer's **GLSL** options in its settings interface in order for it to work.  Instructions for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

laracb01, 2024-02-19:
Thanks! I solved this problem. However, I am unable to see both datas combined. Meaning, I can't combine the receiving image from my REALSENSE d435 and my LIDAR L515. When I click on SOURCE, it only allows me to see one of them, not the combination of both in order to get a better understanding of my surroundings

MartyG-RealSense, 2024-02-19:
If two cameras are attached then Add Source will only list one of the two, and the other will be automatically listed in the side-panel of the Viewer when it is detected.  Once you select the second camera with Add Source then both will be listed in the side-panel and you can select the options of each individual camera.

MartyG-RealSense, 2024-02-27:
Hi @laracb01  Do you require further assistance with this case, please?  Thanks!

laracb01, 2024-02-27:
Yes! I was wondering if you could help me connect my L515 and my D435 in m
y Visual Code Studio in order to extract the capturing data.

Missatge de MartyG-RealSense ***@***.***> del dia dt., 27 de
febr. 2024 a les 11:39:

MartyG-RealSense, 2024-02-27:
If you have installed the full RealSense SDK on Windows using the Intel.RealSense.SDK-WIN10 installer program then you can use the guide at the link below to link a Visual Studio project to the RealSense libraries using .props property sheet files provided by the SDK installation.

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

laracb01, 2024-02-27:
I am trying to do all of this on my Visual Studio Code. I added all the
.props files however, when using the following script:
https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/python-tutorial-1-depth.py
the following message appears: Frame didn't arrive within 5000.
When accesing to my L515 with the Intel Real Sense Viewer, I am able to see
all the images but, however, when working on the Visual Studio COde, I can't

Missatge de MartyG-RealSense ***@***.***> del dia dt., 27 de
febr. 2024 a les 12:19:

MartyG-RealSense, 2024-02-27:
Is there any difference if you unplug the micro-sized end of the L515's USB cable from the base of the camera, reverse the connector's orientation and plug it back into the L515 camera, please (USB-C cables are two-way insertion at the micro-sized end).

laracb01, 2024-02-27:
Well, I am using the following code in order to ensure that the computer
detects the L515:

import pyrealsense2 as rs

def check_realsense_connected():
    ctx = rs.context()
    devices = ctx.query_devices()
    realsense_found = False

    for dev in devices:
        if dev.supports(rs.camera_info.name):
            print("Intel RealSense camera found:", dev.get_info(
rs.camera_info.name))
            realsense_found = True

    if not realsense_found:
        print("No Intel RealSense camera found.")

if __name__ == "__main__":
    check_realsense_connected()

And when the camera is connected the right message appears. Same thing
happens when the camera is not connected.

Missatge de MartyG-RealSense ***@***.***> del dia dt., 27 de
febr. 2024 a les 15:15:

MartyG-RealSense, 2024-02-27:
Thanks very much for providing your script.

The reason for testing the L515 with a reversed orientation of the micro-sized USB connector is that the L515 model typically performs better when the connector is inserted in one particular direction of the two insertion directions available.  So the camera will still pass the detection test if the connector is not inserted in the best direction but may have lower performance.

laracb01, 2024-02-27:
But my point is the following:  I want to connect my L515 and my D435 in my
Visual Code Studio in order to extract the capturing data.  I am trying to
do all of this on my Visual Studio Code. I added all the .props files
however, when using the following script:
https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/python-tutorial-1-depth.py
the following message appears: Frame didn't arrive within 5000.
When accesing to my L515 with the Intel Real Sense Viewer, I am able to see
all the images but, however, when working on the Visual Studio COde, I
can't extract data

Missatge de MartyG-RealSense ***@***.***> del dia dt., 27 de
febr. 2024 a les 18:41:

MartyG-RealSense, 2024-02-28:
Are both cameras detected and streamed if the **multiple_realsense_cameras.py** Python RealSense multicam script at the link below is run?

https://github.com/ivomarvan/samples_and_experiments/tree/master/Multiple_realsense_cameras

closed 2024-03-13 by MartyG-RealSense