realsenseai/librealsense #12685: L515 Depth stream freezes after some mechanical impact

AlexTiniakov, 2024-02-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.8.1 | 
| Operating System & Version |  Linux (Ubuntu 20.04) | 
| Kernel Version (Linux Only)    |  5.15.0-94-generic                                         | 
| Platform                                 | PC/Pico  |
| SDK Version                            | 2.51.1                         | 
| Language                            |  C++ on ROS/ROS2                          | 
| Segment			|  Hardware                          | 

### Issue Description
Hello, I work with L515 camera for several years. We used ROS1 to launch camera and receive data, such as depth, RGB and IMU data. There is one interesting issue: camera sometimes stops stream depth data, especially after mechanical impact (almost 100% possibility).  
In realsense-viewer you can handle it easily: turn off and on the L500 Depth Sensor
[image]  
In ROS1 there was a special service to reboot camera, that helps: "realsense2_camera/reset"  
But how handle it in ROS2? I tried something like this:  
```
const char* kSetParametersService = "realsense2_camera/set_parameters";

set_parameters_client_ =
    std::make_shared<services::SyncServiceClient<rcl_interfaces::srv::SetParameters>>(
      kSetParametersService, *this);

bool ObstacleDetectionNode::ResetDepthModule()
{
  bool success = false;

  RCLCPP_INFO(get_logger(), "Start ResetDepthModule in ros2");

  for (auto i: {false, true}) {
    auto request = std::make_shared<rcl_interfaces::srv::SetParameters::Request>();

    rcl_interfaces::msg::Parameter parameter;
    parameter.name = kEnableDepthParameterName;

    rcl_interfaces::msg::ParameterValue value;
    value.type = rcl_interfaces::msg::ParameterType::PARAMETER_BOOL;
    value.bool_value = i;

    parameter.value = value;

    request->parameters.push_back(parameter);

    using namespace std::chrono_literals;
    while (!set_parameters_client_->WaitForService(1s)) {
      if (!rclcpp::ok()) {
        RCLCPP_ERROR(
          get_logger(), "SetParameters: Interrupted while waiting for the service. Exiting.");
        exit(-1);
      }
      RCLCPP_INFO(get_logger(), "SetParameters: service not available, waiting again...");
      break ;
    }

    auto response = std::make_shared<rcl_interfaces::srv::SetParameters::Response>();
    if (set_parameters_client_->Call(request, response)) {
      RCLCPP_INFO_STREAM(get_logger(), "Set parameters: " << response->results[0].successful);
      success = response->results[0].successful;
    } else {
      RCLCPP_ERROR(get_logger(), "Failed to call service realsense2_camera/set_parameters");
      success = false;
    }
  }

  return success;
}
```

But `set_parameters_client_->WaitForService(1s)` always return `false`
What mechanism of reset depth stream or reboot camera in ROS2 do you have?

MartyG-RealSense, 2024-02-19:
Hi @AlexTiniakov  There is not a camera reset service for the RealSense ROS2 wrapper that can be used during runtime but you may be able to stop and start the depth stream during runtime with **ros2 param set** commands.

```
ros2 param set /camera/camera enable_depth false
ros2 param set /camera/camera enable_depth true
```

If the problem is occurring when the camera is knocked then you could also investigate whether it is occurring because the USB cable's connector is shifting position in the USB port during the collision.

AlexTiniakov, 2024-02-19:
> Hi @AlexTiniakov There is not a camera reset service for the RealSense ROS2 wrapper that can be used during runtime but you may be able to stop and start the depth stream during runtime with **ros2 [...]

It is 100% not an issue with cable. You can see video demo of this issue in my youtube channel by this [link](https://youtu.be/LsGAFyFlPxg?si=lXCHJzfFBijsZSKM)

MartyG-RealSense, 2024-02-19:
The RealSense Viewer has a known issue where enabling three streams (Depth, RGB, Motion Module) can cause the depth or RGB stream to become No Frames Received.  There is not a fix for this issue in the Viewer except to only enable two streams (such as depth + RGB or depth + IMU).  Enabling the Motion Module is the trigger for the issue, so changing the order that the streams are enabled in does not prevent it from occurring.

Does the problem still occur if you only enable Depth and RGB and leave the Motion Module disabled?

AlexTiniakov, 2024-02-19:
> The RealSense Viewer has a known issue for some RealSense users where enabling three streams (Depth, RGB, Motion Module) can cause the depth or RGB stream to become No Frames Received. There is not a fix for this issue in the Viewer except to only enable two streams (such as depth + RGB or depth + IMU). Enabling the Motion Module is the trigger for the issue when three streams become active, so changing the order that the streams are enabled in does not prevent it from occurring.
> 
> Does the problem still occur if you only enable Depth and RGB and leave the Motion Module disabled?

With enable only Depth and RGB the problem still occur

AlexTiniakov, 2024-02-19:
> Hi @AlexTiniakov There is not a camera reset service for the RealSense ROS2 wrapper that can be used during runtime but you may be able to stop and start the depth stream during runtime with **ros2 [...]

It works! Thanks. Woh can I set parameters from c++ code? Now code freezes when I call the service "realsense_camera/set_parameters" at the line of code above:
`if (set_parameters_client_->Call(request, response)) {`

MartyG-RealSense, 2024-02-19:
It is possible to write a C++ **node script** for ROS2 to initiate C++ code from ROS.

https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Using-Parameters-In-A-Class-CPP.html

The RealSense ROS wrapper only has example node scripts for Python though.

https://github.com/IntelRealSense/realsense-ros/tree/ros2-development/realsense2_camera/scripts

In a non-ROS RealSense C++ script, you can stop the pipeline and then start it again.

AlexTiniakov, 2024-02-20:
> It is possible to write a C++ **node script** for ROS2 to initiate C++ code from ROS.
>
> https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Using-Parameters-In-A-Class-CPP.html
>
> The [...]

In some reason SyncServiceClient freezes and never return, but works AsyncServiceClient:

```
auto srv_client = services::AsyncServiceClient<rcl_interfaces::srv::SetParameters>(
      kSetParametersService, *this);
...
  success = srv_client.Call(
    request, [&](auto /*res*/) { RCLCPP_INFO_STREAM(get_logger(), "Set parameters: Success"); });
```

closed 2024-02-20 by AlexTiniakov

MartyG-RealSense, 2024-02-20:
Thanks very much for sharing your AsyncServiceClient solution!