realsenseai/librealsense #12690: Motion Module when off cannot be turned back on.

micmercer1, 2024-02-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Ubuntu 18.04.6 | 
| Kernel Version (Linux Only)    |  4.9.337   | 
| Platform                                 | NVidia Jetson Nano  |
| SDK Version                            |  2.54.2.0 (maybe)       | 

### Issue Description
When attempting to view the IMU data for the D435i it turns on and seems to work fine until the toggle is switched off. Then when I attempt to flip the toggle back on I get one of two errors and it does not turn on or show the coordinate system. Seems like the stereo and RGB are working fine.

1) Backend in rs2_stop(sensor:0x7f9001a1b0):
Failed to enable_sensor
/sys/devices/70090000.xusb/usb2/2-1/2-1.3/2-1.3:1.5/0003:8086:0B3A.0003/HID-SENSOR-2000e1.3.auto/enable_sensor
Last Error: Permission denied

or

2) Wrong Api Call Sequence in rs2_open_multiple(sensor:0x7f9001a1b0,
profiles:0x7f904925d8, count:2):
open(...) failed. Hid device is streaming!

micmercer1, 2024-02-21:
Hi Marty! 

This worked! For future notes, I think the problem was stemming from trying to have the realsense-viewer open while also trying to export IMU data for our controls algorithm using Pyrealsense2. Once I did the hardware reset it worked, but I noticed the error popped up again when I was using Pyrealsense2. I did the hardware reset and closed the script and it was back to working just fine.

micmercer1, 2024-02-21:
Thank you for the help!

MartyG-RealSense, 2024-02-21:
You are very welcome.  I'm pleased to hear that you had a good outcome.  Thanks very much for the update!