realsenseai/librealsense #12691: 3d recording

Hassancfc, 2024-02-21:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f1181691|
|**Firmware Version**|1.5.8.1|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003c0144405|
|**Firmware Update Id**|0003c0144405|
|**Name**|Intel RealSense L515|
|**Serial Number**|f1181691|
|**Firmware Version**|1.5.8.1|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003c0144405|
|**Firmware Update Id**|0003c0144405|

Hi, i want to record a 3d point cloud( if the term is right) but when i connect the camera it show the objects for 2 seconds than everything disappear. it than only shows the camera in the middle of the screen

MartyG-RealSense, 2024-02-21:
Hi @Hassancfc  Are you using the RealSense Viewer tool, please?  If you are then this effect can be caused by a setting called **GLSL**.  https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023 has instructions for how to disable GLSL in the Viewer to stop the pointcloud from disappearing.

Hassancfc, 2024-02-25:
thanks MatyG, i just started with lidar and point clouds and i have a question regarding post processing a point cloud, like crop, highlighting important area. do you have any recommendation

closed 2024-02-25 by Hassancfc

MartyG-RealSense, 2024-02-25:
The L515 camera model does not support using post-processing to set a minimum and maximum distance crop for the depth image in the Viewer unfortunately.  That is a feature of the 400 Series stereo depth cameras.

An alternative method on L515 for setting a minimum distance for depth rendering is to use the **Min Distance** option.  Depth values below the user-defined minimum distance will not be rendered on the depth image.

[image]

You can define a **Region of Interest** (ROI), within which pixels will be maintained at an intensity controlled by a user-defined **setpoint** value.  More information about controlling the ROI area and setpoint in the Viewer can be found in point (3) of the link below.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#make-sure-image-is-properly-exposed