realsenseai/librealsense #12695: UNKNOWN in rs2_get_option(options:000002ECCF861D00, option:Backlight Compensation): hr returned: HResult 0x80070490: "Element not found.

Fau786, 2024-02-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 }/435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Greetings
I have recently started with my project using 435i version.As i connected to my PC.the depth quality tool gives me both 2D and 3d VIEW but for viewr v2.54.2 window niether show frames nor showing rgb icon on side.i have updated my firmware using previous post issue but of no avail.
how could i solve my problem.as i am not geting frames on viewr but on depth tool.
Regards

MartyG-RealSense, 2024-02-22:
Hi @Fau786  You mention that the RGB icon is not showing in the Viewer's side-panel.  Does the camera name at the top of the side-panel say D435i or **D430i** please.

Fau786, 2024-02-22:
its showing as D435.

MartyG-RealSense, 2024-02-22:
Thank you.  Which firmware version have you installed?

Fau786, 2024-02-22:
i just follow the link to install and down fron this link 
https://github.com/IntelRealSense/librealsense/releases/download/v2.54.2/Intel.RealSense.SDK-WIN10-2.54.2.5684.exe

MartyG-RealSense, 2024-02-22:
Okay, thank you.  The firmware driver is not updated by the Intel.RealSense.SD-WIN10 installer program and has to be updated separately.

Please click the 'Info' button at the top of the Viewer's side-panel and see which version number is listed for the firmware.

Fau786, 2024-02-22:
its 5.15.1

MartyG-RealSense, 2024-02-22:
A D435i camera should not be listed as a D435.  If it is then it would suggest the camera's Motion Module IMU component cannot be detected.

Is there a Motion Module section in the side-panel or does it only list Stereo Module.

Fau786, 2024-02-22:
it is written as stereo model

Fau786, 2024-02-22:
i have plug unplug the USB port.now it is showing RGB camera and stereo.but no frames received is still on screen for viewer

MartyG-RealSense, 2024-02-22:
Next please try resetting the calibration of your camera to its factory-new default settings in the Viewer using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 to see whether it corrects the problems that you are experiencing.

Fau786, 2024-02-22:
i will see it also i want to start with some python examples.how could i start with this.anygithub repository.
regards

MartyG-RealSense, 2024-02-22:
The RealSense SDK has some Python examples at https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples

A starter example is here:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#examples

Fau786, 2024-02-23:
hey.
No still i am not able to receive any frame on my viewer window but again on depth quality side?
how i could solve this issue?

MartyG-RealSense, 2024-02-23:
Would it be possible to post an image of your Viewer window into a comment please?

Fau786, 2024-02-23:
so this is the main issue.secondly frames arrive too late.where everythig went wrong for me?

MartyG-RealSense, 2024-02-23:
The images show that the camera is being detected as the **D435** model (depth and RGB but not IMU) and that it is being detected as being on a slow USB 2 connection instead of the faster USB 3.

Is your camera plugged into a USB 2 port or a USB 3 port?

If it is in a USB 3 port, are you using the official 1 meter USB cable supplied with the camera or a cable of your own choice?  If it is your own choice of cable then it might be a USB 2 cable that would prevent the camera from operating at its full capability.

If the camera was plugged into a USB 2 hub instead of a USB 3 hub then that would also limit the camera's capabilities.

Fau786, 2024-02-23:
the camera is plugged into USB 3 port but i am not sure about the cable(i dont think so its an official cable)provided to me by supervisor.
so the reason of error lies in my cable?

MartyG-RealSense, 2024-02-23:
It is easy to mistakenly use a USB 2 cable instead of a USB 3 one as they look almost the same.  One way to identify a USB3 cable is to see if the letters **SS** (which stands for superspeed) are printed on the connector.

Fau786, 2024-02-23:
the device is connected to USB 3 as i connect the device it the frames start to show on depth quality tool but for viewer window its still say no frames received.

MartyG-RealSense, 2024-02-23:
Even if a camera is connected to a USB 3 port, it will operate at USB 2 speed if the cable is USB2.  This is because USB3 cables have some extra wires inside them that allow USB3 mode to be enabled.

Fau786, 2024-02-23:
so firstly should i change my cable then to analyze the difference.
i will continue to this discussion once i changed the cable.

MartyG-RealSense, 2024-02-23:
I look forward to your next update.  Good luck!

MartyG-RealSense, 2024-03-08:
Hi @Fau786   Do you require further assistance with this case, please?  Thanks!

Fau786, 2024-03-08:
Hey.
yesterday i received my cable.and yes the issue of getting frames solved.
Thank you so much for follow up
Regards

MartyG-RealSense, 2024-03-08:
You are very welcome.  It's great to hear that a new cable resolved your issue.  Thanks very much too for the update!

closed 2024-03-08 by MartyG-RealSense