realsenseai/librealsense #12697: Realsense2 program hangs when calling pipeline.stop()

Westluu, 2024-02-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 5.15.1.0 | 
| Operating System & Version |  Ubuntu Desktop 22.04 LTS | 
| Kernel Version (Linux Only)    |  5.15.0-1023-xilinx-zynqmp                                      | 
| Platform                                 | Kria KR26  |
| SDK Version                            |  realsense2                          | 
| Language                            |  python   | 
| Segment			|  Camera                     | 

### Issue Description
```
import pyrealsense2 as rs
import numpy as np
import cv2
import time
import torch
import math

# MAIN

import signal
import sys

# program status
RUNNING = True

# Handle sigint
def signal_handler(signal, frame):
    global RUNNING
    print("Received sigint")
    RUNNING = False

signal.signal(signal.SIGINT, signal_handler)

# Set up pipeline
pipeline = rs.pipeline()
config = rs.config()

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
pipeline.start(config)
i = 0
try:
    while RUNNING:
         #NOP

finally:
    print("finally")
    print(RUNNING)
    pipeline.stop()
    print("PIPILINE STOP")
    exit()
```

A problem occurs when I try to exit the program or call pipeline.stop(), the program hangs and creates a bunch of child processes in state S. 

When I try to kill them, it then creates a process in state D causing the program to hang forever.

Previously when using the jetson nano this problem was not an issue.

But when switching to the KRIA KV260 board using the python wrapper of realsense 2 this problem occurs.

Installed realsense2 by getting the source files and compiling them on the Kria Board following these instructions:
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#readme

MartyG-RealSense, 2024-02-23:
Hi @Westluu  As the **KRIA KV260** board uses **xilinx** Linux instead of Ubuntu, the best chance of achieving a working pyrealsense2 setup may be to follow the procedure below.

1.  Compile librealsense from source code with CMake with the flag **-DFORCE_RSUSB_BACKEND=ON** included in the CMake build instruction.  An RSUSB = true build of the SDK is useful for 'exotic' computing devices such as industrial boards as it is not dependent on Linux versions or kernel versions and does not have to have a kernel patch applied to it.

2.  As your board uses Arm processor architecture, if you are using Python 3.7, 3.8 or 3.9 and are using librealsense 2.54.1 or newer then you could quickly and easily install the pyrealsense2 wrapper with the command **pip install pyrealsense2**

If your Python version is newer than 3.9 then you should install the wrapper from source code.

MartyG-RealSense, 2024-02-24:
It's great to hear that you were successful, @Westluu - thanks very much for the update!

closed 2024-02-25 by Westluu