realsenseai/librealsense #12698: Cannot change resolution

asuka-ms, 2024-02-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | - | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |  -  | 
| Platform                                 | PC  |
| SDK Version                            |  legacy / 2.54.2  | 
| Language                            |  unity   | 
| Segment			|  others  | 

### Issue Description
I have tried several samples in Unity and they worked well.
However, the Color and Depth resolutions are set to 640*480, and the data generated looks very coarse.
So I changed the resolution in the Inspector of RsDevice.
Then I got an error like the one below.

--------------------------------------------------------------
ExternalException: rs2_pipeline_start_with_config(pipe:00000269AF56F180, config:00000269AF06DBC0)
Rethrow as Exception: Couldn't resolve requests
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <174f4594684e480d948f375945ed1b21>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <174f4594684e480d948f375945ed1b21>:0)
RsDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)
--------------------------------------------------------------

I searched the net and discovered that it apparently has to be the same as the value of RsVideoStreamRequest.
I changed RsDevice.cs to look like the image below but could not get it to run with the same error.
The resolution of D435i is 1920*1080 for Color and 1280*720 for Depth, so both are aligned to 1280*720.

[image]

I would like to know how to solve this error or how to change the resolution.

MartyG-RealSense, 2024-02-23:
Hi @asuka-ms  **Couldn't resolve requests** indicates that you may have requested a resolution / FPS setting that was not supported by the camera.  

For example, if you had requested 30 FPS for 1280x720 and the camera was being detected as being on a **USB2** connection instead of USB3 then only 6 FPS would be supported for 1280x720 by your D435i on USB2, whilst 30 FPS would be supported on a USB3 connection for that resolution.

asuka-ms, 2024-02-27:
Thanks for the reply.
I connected it with a USB 3.0 cable and it worked fine!

MartyG-RealSense, 2024-02-27:
I'm pleased to hear that a USB 3 cable resolved your issue.  Thanks very much for the update!

closed 2024-03-06 by MartyG-RealSense