realsenseai/librealsense #12701: D435i doesnt work

pgluz, 2024-02-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |  Ubuntu 22.04 LTS  | 
| Kernel Version (Linux Only)    |  6.5, 5.15                                         | 
| Platform                                 | PC |
| SDK Version                            | 2.54.2                          | 

I have a serious problem when trying to install the D435i on Ubuntu 22.04. I've installed the SDK according to [these instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages).
The camera is recognised as a USB device and found by realsense-viewer, but neither RGB nor stereo image is detected. The viewer shows only "Frames didn't arrived within 5 seconds!". Also, modinfo uvcvideo does not include "realsense" string, as mentioned in the instructions.

I've tried on kernels 6.5 and 5.15. I also tried to patch 5.15 kernel with [patch-realsense-ubuntu-lts-hwe.sh](https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-lts-hwe.sh) without effects, just output "Only garbage was found in the patch input".

The camera itself definitely works because I've tested it on Realsense Viewer on Windows 11. I have no idea what's going on with this Linux, so I will be extremely grateful for any help.

I've found similar problems on GitHub from the past but without any specific solution.

MartyG-RealSense, 2024-02-24:
Hi @pgluz  Support for kernel 6.5 has been added to the **development** branch of librealsense, which becomes the next master branch of the SDK once released.  The next librealsense master release is anticipated to be during the Q1 2024 window (between January and March).

https://github.com/IntelRealSense/librealsense/tree/development

The current librealsense master (2.54.2) supports kernel 5.15.  However, a kernel patch script should not be applied to a DKMS package installation of librealsense (which is what the instructions that you linked to are) as the patch is already bundled within the packages.

The absence of the  "realsense" string when using recent kernel versions has previously been reported by a couple of RealSense users - https://github.com/IntelRealSense/librealsense/issues/12581 is an example of such a case - but it does not appear to prevent librealsense from working normally.

I would recommend first removing your previous package installation of librealsense from your computer using the instruction below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

Then try building librealsense from source code using the build script at the link below.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

This installation method bypasses the Linux kernel and so is useful for getting librealsense working on computers that are having kernel related problems.  It also does not require the application of a patch script.

padhupradheep, 2024-02-29:
>Then try building librealsense from source code using the build script at the link below.
https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md
This installation method bypasses the Linux kernel and so is useful for getting librealsense working on computers that are having kernel related conflicts with librealsense. It also does not require the application of a patch script.

Just running this should take care of all the necessary flags right?

MartyG-RealSense, 2024-02-29:
The **libuvc_installation.sh** build script will build the SDK in kernel bypass mode, though it will not include the example programs and tools such as realsense-viewer.  So if you need those tools and examples then you could create your own custom version of the build script (at the link below) and edit its CMake build instruction.

https://github.com/IntelRealSense/librealsense/raw/master/scripts/libuvc_installation.sh

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=TRUE -DBUILD_GRAPHICAL_EXAMPLES=TRUE`

padhupradheep, 2024-02-29:
Thanks.. this is some really useful info..

MartyG-RealSense, 2024-03-16:
Case closed due to no further comments received.

closed 2024-03-16 by MartyG-RealSense

padhupradheep, 2024-03-18:
Sorry for the late response. Even the bypass solution does not work actually as expected. Causes a lot of other problems.