realsenseai/librealsense #12702: Realsense-viewer crashes after displayng message "realsense device detected"

FeliciaSeq, 2024-02-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu20.04}
### Issue Description

Hi,

Im connected to 5 realsense cameras and when I run realsense viewer, it crashes with a terminal error :
25/02 15:28:58,372 WARNING [139966927591168] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
RealSense error calling rs2_supports_device_info(dev:0x7f4ca0000b60, info:14):
    invalid enum value for argument "info"

MartyG-RealSense, 2024-02-26:
Hi @FeliciaSeq  Does the RealSense Viewer still crash if only one camera is attached, please?  

The more cameras that are attached to a computer, the more of its resources that are consumed.  So a computer with a CPU equivalent to an Intel Core i7 or better is recommendable if all of the cameras will be enabled at the same time.

FeliciaSeq, 2024-02-27:
Hey Marty,

We can simultaneously run 5 cameras at a time with ROS without any issues. On top of that we have an 18-core industrial CPU.

MartyG-RealSense, 2024-02-28:
Thanks very much for the information.  Is there any difference if you launch the Viewer in sudo admin permissions mode with the command **sudo realsense-viewer**

closed 2024-03-13 by MartyG-RealSense